letscontrolit/ESPEasy #812: JSON (plugin?) for parsing MQTT commands

Grovkillen, 2018-02-02:
I wish I could send my units JSON strings using MQTT. If I then could intercept them the same way that MQTT import works today but instead of using a specific topic I could RegEx or even better JSON parse the value I wanted to get. That way a command could look something like this:

topic
`<unit name>/cmd`

payload
```
json,{"command": {
  "reboot": 1,
  "unit": 33,
   "TimerSet": [
      {"Timer1": 10},
      {"Timer2": 0}
  }
}}
```

Or in a single line: 
`json,{"command": {"reboot": 1,"unit": 33,"TimerSet": [{"Timer1": 10},{"Timer2": 0}}}}`

In the device (named "command") I could then have a setting called "reboot" which would then result in the value of "1" given the json string above.

Having this rule would then reboot the unit:
```
On Command#Reboot=1 Do
 reboot
Endon
```

Another setting called "TimerSet.Timer1" would result in a value of "10".
```
On Command#TimerSet.Timer1 Do
 TimerSet,1,[Command#TimerSet.Timer1]
Endon
```
I guess the rules isn't needed if we fully implement this, maybe not even a plugin is needed if we could have it in the core code?

To wrap it up. Don't know if this is a possible plugin to create but given this we could stick with the values only for variables but give them more intelligence and easier to create powerful commands over MQTT. I guess also possible to send strings with this, to be able to rename the device remotely using MQTT?

TD-er, 2018-02-02:
What should listen to these MQTT commands?
The MQTT import plugin, or can you also do it via a controller?

Grovkillen, 2018-02-02:
Well the controller is already listening to the `<unit name>/cmd` topic. So no extra needed...?

TD-er, 2018-02-02:
OK, sounds like something that might not be that complicated to implement

Grovkillen, 2018-03-25:
https://www.letscontrolit.com/forum/viewtopic.php?t=5096

An example of where a generic json parser for MQTT would be of use.

BugsonPL, 2019-09-23:
Considering more and more devices use JSON MQTT messages, JSON parser definitely should be part of  EasyESP. Maybe as a extension to existing EasyFormula ?