letscontrolit/ESPEasy #814: NTP unreliable

s0170071, 2018-02-02:
2.4.0 mega. cant reproduce it yet but now and again the year is 2036 :-O
If I could get it to be 2036 reliably, I would hook it up to flux capacitor of my car.

```
INIT : Booting version: (custom)
INIT : Warm boot #33
FS   : Mounting...
FS   : Mount successful, used 76053 bytes of 957314
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:22000
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Connecting MNET attempt #1
Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
WIFI : Connected! IP: 192.168.1.32 (ESP_Easy_11)
Current Time Zone:  DST time start: 2036-03-30 02:00:00 offset: 120 minSTD time start: 2036-10-26 03:00:00 offset: 60 min
WD   : Uptime 0 ConnectFailures 0 FreeMem 18400
s$
```

s0170071, 2018-02-02:
another one: 
```
  : SecuritySettings CRC   ...OK 
INIT : Free RAM:22000
INIT : I2C
INIT : SPI not enabled
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
pm open,type
connected with MNET, channel 6
dhcp client start...
:2 0
mode : sta(5c:cf:7f:39:54:8d)
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Connecting MNET attempt #1
sl
scandone
usl
ip:192.168.1.32,mask:255.255.255.0,gw:192.168.1.1
NTP  : NTP sync request:1
NTP  : NTP send to 138.68.46.177
NTP  : No reply
NTP  : NTP sync request:2
NTP  : NTP send to 138.68.46.177
NTP  : No reply
NTP  : NTP sync request:3
NTP  : NTP send to 138.68.46.177
NTP  : No reply
WIFI : Connected! IP: 192.168.1.32 (ESP_Easy_11)
NTP  : NTP sync request:1
NTP  : NTP send to 138.68.46.177
NTP  : No reply
NTP  : NTP sync request:2
NTP  : NTP send to 138.68.46.177
NTP  : No reply
NTP  : NTP sync request:3
NTP  : NTP send to 138.68.46.177
NTP  : No reply
WD   : Uptime 0 ConnectFailures 0 FreeMem 18400
WD   : Uptime 1 ConnectFailures 0 FreeMem 18400
HTML : Template-Size=340 Content-Size=414 [1913] [471] [414] [72] [0]
HTML : Request file /favicon.ico
WD   : Uptime 1 ConnectFailures 0 FreeMem 17928
HTML : Request file /favicon.ico

```

The web interface then reports 1970.

TD-er, 2018-02-02:
Sounds like "-1" is being interpreted as valid time response from the NTP server.

s0170071, 2018-02-02:
and one more from 2036...

```
NIT : Booting version: (custom)
INIT : Warm boot #24
FS   : Mounting...
FS   : Mount successful, used 76053 bytes of 957314
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:21872
INIT : I2C
INIT : SPI not enabled
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
pm open,type:
connected with MNET, channel 6
dhcp client start...
2 0
mode : sta(5c:cf:7f:39:54:8d)
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Connecting MNET attempt #1
sl
scandone
usl
ip:192.168.1.32,mask:255.255.255.0,gw:192.168.1.1
NTP  : NTP sync request:1
NTP  : NTP send to 192.33.214.57
NTP  : No reply
NTP  : NTP sync request:2
NTP  : NTP send to 192.33.214.57
NTP  : NTP replied: 47 mSec
Current Time Zone:  DST time start: 2036-03-30 02:00:00 offset: 120 minSTD time start: 2036-10-26 03:00:00 offset: 60 min
WIFI : Connected! IP: 192.168.1.32 (ESP_Easy_11)
NTP  : NTP sync request:1
NTP  : NTP send to 192.33.214.57
NTP  : No reply
NTP  : NTP sync request:2
NTP  : NTP send to 192.33.214.57
NTP  : No reply
NTP  : NTP sync request:3
NTP  : NTP send to 192.33.214.57
NTP  : No reply
WD   : Uptime 0 ConnectFailures 0 FreeMem 18272

```

s0170071, 2018-02-02:
if you try hard enough, you don't get any answers from the ntp for a while. In that case no time is displayed in the serial console.
```

INIT : Warm boot #5
FS   : Mounting...
FS   : Mount successful, used 76053 bytes of 957314
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:21872
INIT : I2C
INIT : SPI not enabled
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
pm open,type
connected with MNET, channel 6
dhcp client start...
:2 0
mode : sta(5c:cf:7f:39:54:8d)
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Connecting MNET attempt #1
sl
scandone
usl
ip:192.168.1.32,mask:255.255.255.0,gw:192.168.1.1
NTP  : NTP sync request:1
NTP  : NTP send to 91.224.149.41
NTP  : No reply
NTP  : NTP sync request:2
NTP  : NTP send to 91.224.149.41
NTP  : No reply
NTP  : NTP sync request:3
NTP  : NTP send to 91.224.149.41
NTP  : No reply
WIFI : Connected! IP: 192.168.1.32 (ESP_Easy_11)
NTP  : NTP sync request:1
NTP  : NTP send to 91.224.149.41
NTP  : No reply
NTP  : NTP sync request:2
NTP  : NTP send to 91.224.149.41
NTP  : No reply
NTP  : NTP sync request:3
NTP  : NTP send to 91.224.149.41
NTP  : No reply
WD   : Uptime 0 ConnectFailures 0 FreeMem 18272

```

TD-er, 2018-02-02:
2036 is the new millennium problem.
That's 2^32 - 1 seconds from Unix epoch. (1970-01-01)

So the time was set to "-1" which is 2^32-1 for an unsigned int.
1 second later, it is 1970 again ;)

s0170071, 2018-02-02:
sigh. Not again. I need to adjust my flux capacitor for this.

TD-er, 2018-02-27:
NTP has now a better query/re-query approach. But still it is possible to have a timezone setting west from Greenwich that will set the NTP to negative values (0 is the error value of NTP)... ehh I mean those timezones will activate the flux capacitor.

So until that has to be fixed, this issue can be closed.
If the ESP needs to support proper time travelling, please open an new issue. (preferably in the future)

closed 2018-02-27 by TD-er