letscontrolit/ESPEasy #816: Sonoff Basic not responding with ESPEasy 2.0 20180124 or later

wdonker, 2018-02-02:
Some days ago I decided to 'upgrade' my Sonoff Basic from v2.0-20180112 to v2.0-20180130 and found that the webpages are no longer responding. Also the serial log shows it is losing the MQTT connection (I use Domoticz MQTT controller).

I tried several nightlies and v2.0-20180123 is the last build that works ok; all the later ones are not working well. To track down the problem I compiled ESPEasy from source 2.0-201830 but replaced P036-plugin with the version of 20180123 (since this plugin was one of the changes since 20180123). This customized version runs ok, so the problem seems to be in P036.

Some other background information:
- v2.0-2101830 runs fine on my NodeMCU's, only the Sonoff is having problems.
- there is no P036 device in my config so it is strange the P036 plugin can cause a problem.
- disabling the MQTT controller did not help.

[image]
[image]

TD-er, 2018-02-02:
OK, will have a look at it.

TD-er, 2018-02-02:
Just a question, does it also happen with the Mega branch, around the same release builds?

The OLED Framed plugin has changed a lot between v2.0 and Mega branch, but those changes are already months ago.

TD-er, 2018-02-02:
This fix was introduced with the build of 24/1:
fix #718. also update display on save and removed graphical boot logo to conserve space. also fixes #713. (web interface will be non repsonsive during scrolling now, but there is no easy fix for that)

wdonker, 2018-02-02:
Yes, I saw it was changed. That's why I tried with the 23/1 source. Strange thing is that I do not use a P036 device so how can this plugin create problems? And only on Sonoff but not on NodeMCU.
I never used mega branch so I don't know if it happens there too. I can try but will take a few days.
For me it is not a big problem now so if nobody else has this problem priority can be low.

TD-er, 2018-02-02:
I will have to look into the code. Let's hope my mind is fresh enough ;)
I have seen other plugins cause similar issues, especially when the devices are not present.
But yes, it is technically possible to cause interference.

wdonker, 2018-02-06:
I've flashed the Sonoff with v2.0-20180206 because I saw some fixes in P036 and now the problem has dissappeared. So I think we can close this issue.

closed 2018-02-06 by wdonker

TD-er, 2018-02-06:
Nice to see it is fixed now :)

jannnfe, 2018-10-28:
Hey @TD-er,
I'm not sure if my problem has anything to do with it, but it all sounds very similar. I use a much newer version something around 20180920 - 20181010. I can not say exactly the version at the moment, because I first have to unscrew my wall socket to take the Sonoff from power, because he does not react anymore. The problem I had already 3-4 times and I thought it may be because of WIFI but all my other devices (like Wemos D1 Mini) with OLED Framed Plugin are working.

So I use my Sonoff Basic in a wall socket to control my heating. An OLED display is connected to the Sonoff. As soon as I no longer get connected to the sonoff, the display stops scrolling and freezes. After about 30 seconds, he flips through all the pages of the plugin in a fast pass. I also recorded a video about it. As soon as I can reconnect to my Sonoff, I will post the version number.

Thanks and Greetings

TD-er, 2018-10-28:
It may be related to some other issue.
I tried to look for it yesterday, but couldn't find it.

In short, at some moment something has happened which causes the network stack to get into some undefined state.
I have not yet a clue what happens, but it following symptoms may be related:
- IP address is said to be 0.0.0.0
- MQTT, or webserver (or both) may not respond anymore

It looks like some of the services are not restarted, or the rest of the services which are already killed are waiting forever to restart.

I saw it yesterday on one of my own nodes, which happens to be mounted outside.

jannnfe, 2018-11-28:
Hello, I still have the same version as before installed and nothing else is set. Since a few weeks, the OLED display continues to run normally and you do not even notice that the WLAN connection is interrupted. I always have to look in my FritzBox, if the Sonoff is still connected. I can only reconnect to my Sonoff when I take it off the electricity. The WPA AP mode does not work either, because no matter what I save under WPA AP Mode Key, my computer tells me that I use a wrong password to connect to the Sonoff via access point.

Do you think updating to newer version will fix the Wifi Problem?

jannnfe, 2018-11-28:
> Edit: Maybe you can start a new issue about this, with all information collected in 1 post.
> That makes it easier to handle it and then I don't have to re-read a lot when it is popping up again.

I have opened a [new issue](https://github.com/letscontrolit/ESPEasy/issues/2098#issue-385212035) and try to get everything together as well as possible.