letscontrolit/ESPEasy #817: Email notification misses sender address

wdonker, 2018-02-02:
### Steps to reproduce
Add email notification with sender and receiver address.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Receiver should see the email adress that is set as sender in ESPEasy.

### Actual behavior
Sender is left blank, causing mailservers like gmail to refuse the email.

### System configuration

Hardware:
NodeMCU with nothing connected.

Software or git version:
v2.0-20180124

TD-er, 2018-02-03:
Can you add some screenshot and log? (part of email address can be marked black)
And it struck me there is no password field. Are there any mail providers left that accept sending mail without passwd?

TD-er, 2018-02-03:
Ah found it, the mail header presented is just incomplete. (just like you described)

wdonker, 2018-02-03:
I also expected gmail to require a password but it was returned as undeliverable because of the missing from addresss.
Maybe the next issue is the lack of  user/pwd ;-)
Thnx for the fix.

TD-er, 2018-02-03:
Could you test this pull req.?

I was not yet able to test it here and am now looking into username/passwd authentication for email so we can use Gmail.

wdonker, 2018-02-03:
Yes, it is fixed. From unit name ESP8266-5 the mailheader now contains:

**From: ESP8266-5 <espeasy@xxx.nl>**

TD-er, 2018-02-03:
Cool :)
I am still looking into adding username/passwd so we can also use it with Gmail.
Not sure how long it will work, since most mail providers demand TLS/SSL and that's a bit much.

wdonker, 2018-02-03:
Ah, maybe Gmail already does?

I tested by sending the notification to my local mailserver which auto-forwards to my gmail address.
Without the sender address the mail was not accepted by Gmail; now it is.

TD-er, 2018-02-04:
Added authentication (tested with smtp2go.com)

wdonker, 2018-02-06:
I've tested with gmail server but indeed that needs SSL/TLS.
No problem, but it revealed a little error in the logging of N001 (missing ':'):
EMAIL: Connecting to smtp.gmail.com25
EMAIL: Error connecting to smtp.gmail.com25

TD-er, 2018-02-06:
That missing colon is just a log entry.
I will correct it if I don't forget :)

Grovkillen, 2018-02-10:
This can be closed.

chaluz74, 2020-05-12:
Hello!
Is there any form of debugging in the smtp notification?
I can't send even using smtp2go.

Git Build: | mega-20200426
991671: EMAIL: Connecting to mail.smtp2go.com2525
991805: EMAIL: Error connecting to mail.smtp2go.com2525

May 11 23:08:21 raspberrypi postfix/qmgr[32133]: E5A54E01FF: removed
May 11 23:13:08 raspberrypi postfix/qmgr[32133]: 0F2C6E01E3: from=<esp001@teste.com>, size=486, nrcpt=1 (queue active)
May 11 23:13:09 raspberrypi postfix/smtp[32204]: 0F2C6E01E3: to=<cha@xxxx.com>, relay=mail.smtp2go.com[66.228.43.14]:2525, delay=172508, delays=172506/0.11/0.75/0.83, dsn=2.0.0, status=sent (250 OK id=1jYLLx-wSEPaG-K0)

I tested the site via raspberrypi and it worked out the problem that smtp2go.com no longer accepts on port 25 and when I put 2525 on EspEasy the email is not sent.

  Can you release port 2525 in the SMTP notficication configuration?

Thanks

uzi18, 2020-05-12:
@chaluz74 if can't connect so maybe timeout is too short we use here 100ms by default:

https://github.com/letscontrolit/ESPEasy/blob/dd4ac70c1128fdad702cb63bd337fca0b2599a56/src/src/DataStructs/ControllerSettingsStruct.h#L47

TD-er, 2020-05-12:
> I tested the site via raspberrypi and it worked out the problem that smtp2go.com no longer accepts on port 25 and when I put 2525 on EspEasy the email is not sent.
> 
> Can you release port 2525 in the SMTP notficication configuration?

@chaluz74 Can you make a new issue for that?
It is a feature request and those are impossible to find at the end of closed issues.

uzi18, 2020-05-12:
@chaluz74 try this one and give me a log:
[ESP_Easy_mega_20200512_normal_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/4615228/ESP_Easy_mega_20200512_normal_ESP8266_4M1M.zip)

chaluz74, 2020-05-12:
Hello!
Is there any form of debugging in the smtp notification?
I can't send even using smtp2go.

Git Build: | mega-20200426
991671: EMAIL: Connecting to mail.smtp2go.com2525
991805: EMAIL: Error connecting to mail.smtp2go.com2525

May 11 23:08:21 raspberrypi postfix/qmgr[32133]: E5A54E01FF: removed
May 11 23:13:08 raspberrypi postfix/qmgr[32133]: 0F2C6E01E3: from=<esp001@teste.com>, size=486, nrcpt=1 (queue active)
May 11 23:13:09 raspberrypi postfix/smtp[32204]: 0F2C6E01E3: to=<cha@xxxx.com>, relay=mail.smtp2go.com[66.228.43.14]:2525, delay=172508, delays=172506/0.11/0.75/0.83, dsn=2.0.0, status=sent (250 OK id=1jYLLx-wSEPaG-K0)

I tested the site via raspberrypi and it worked out the problem that smtp2go.com no longer accepts on port 25 and when I put 2525 on EspEasy the email is not sent.

  Can you release port 2525 in the SMTP notficication configuration?

Thank

> 
> 
> @chaluz74 try this one and give me a log:
> [ESP_Easy_mega_20200512_normal_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/4615228/ESP_Easy_mega_20200512_normal_ESP8266_4M1M.zip)

Hi !

Unfortunately I was unable to upload this version, the only one that works here:
ESP_Easy_mega-20200426_normal_ESP8266_1M.bin

I wanted to test HTTP Advanced for Blynk in the test version, but I didn't find it in the ESPEasy_mega-20200426 package. I use ESP01 1M.

If you can send a version with the smtp timeout and HTTP Advanced, thank you very much.

chaluz74, 2020-05-12:
> 
> 
> @chaluz74 try this one and give me a log:
> [ESP_Easy_mega_20200512_normal_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/4615228/ESP_Easy_mega_20200512_normal_ESP8266_4M1M.zip)

Hi !

Unfortunately I was unable to upload this version, the only one that works here:
ESP_Easy_mega-20200426_normal_ESP8266_1M.bin

I wanted to test HTTP Advanced for Blynk in the test version, but I didn't find it in the ESPEasy_mega-20200426 package. I use ESP01 1M.

If you can send a version with the smtp timeout and HTTP Advanced, thank you very much.

uzi18, 2020-05-12:
did you unzipped file before upload?

chaluz74, 2020-05-12:
> 
> 
> did you unzipped file before upload?

Yes, flashing done 131s

[image]

TD-er, 2020-05-12:
> Unfortunately I was unable to upload this version, the only one that works here:
> ESP_Easy_mega-20200426_normal_ESP8266_1M.bin

What you mention is an 1M build and @uzi18 has made a 4M build so it seems (also judging from your screenshot)

uzi18, 2020-05-12:
my bad, one more time: 
[ESP_Easy_mega_20200512_normal_ESP8266_1M.bin.zip](https://github.com/letscontrolit/ESPEasy/files/4618466/ESP_Easy_mega_20200512_normal_ESP8266_1M.bin.zip)

chaluz74, 2020-05-12:
> 
> 
> my bad, one more time:
> [ESP_Easy_mega_20200512_normal_ESP8266_1M.bin.zip](https://github.com/letscontrolit/ESPEasy/files/4618466/ESP_Easy_mega_20200512_normal_ESP8266_1M.bin.zip)
> need error log

It worked !!! follows log .... thank you very much for your help.

If you have the test version I would like to use blynk.

[image]

uzi18, 2020-05-13:
is it possible to have about 10x more of this log?

TD-er, 2020-05-13:
And maybe in public sites don't have the `AUTH LOGIN` part shown.
Even though it looks encrypted, it is probably reversible, so after this, I suggest to change the password on this smtp2go account.

uzi18, 2020-05-13:
please remove this screenshot and change your password on smtp2go, we need only `EMAIL time` lines

@TD-er you see? it is 3x-4x more than standard 100ms

Grovkillen, 2020-05-13:
@uzi18 I edited the message. Good call!

TD-er, 2020-05-13:
> @TD-er you see? it is 3x-4x more than standard 100ms

So the default timeout for SMTP should be maxed to whatever is usable without crashes?
What had you changed in the build?

uzi18, 2020-05-13:
@TD-er this:

```diff
diff --git a/src/_N001_Email.ino b/src/_N001_Email.ino
index 9720f871..785f8155 100644
--- a/src/_N001_Email.ino
+++ b/src/_N001_Email.ino
@@ -77,13 +77,16 @@ boolean NPlugin_001_send(const NotificationSettingsStruct& notificationsettings,
 
        // Use WiFiClient class to create TCP connections
        WiFiClient client;
-       client.setTimeout(CONTROLLER_CLIENTTIMEOUT_DFLT);
+       client.setTimeout(CONTROLLER_CLIENTTIMEOUT_DFLT*10U);
        String aHost = notificationsettings.Server;
-       addLog(LOG_LEVEL_DEBUG, String(F("EMAIL: Connecting to ")) + aHost + notificationsettings.Port);
+       addLog(LOG_LEVEL_DEBUG, String(F("EMAIL: Connecting to ")) + aHost + ':'+ notificationsettings.Port);
+    unsigned long timer1 = millis();
        if (!connectClient(client, aHost.c_str(), notificationsettings.Port)) {
-               addLog(LOG_LEVEL_ERROR, String(F("EMAIL: Error connecting to ")) + aHost + notificationsettings.Port);
+               addLog(LOG_LEVEL_ERROR, String(F("EMAIL: Error connecting to ")) + aHost + ':'+ notificationsettings.Port);
                myStatus = false;
        }else {
+        timer1 = millis()-timer1; 
+               addLog(LOG_LEVEL_ERROR, String(F("EMAIL: time ")) + String(timer1));
                String mailheader = F(
                        "From: $nodename <$emailfrom>\r\n"
                        "To: $ato\r\n"

```

TD-er, 2020-05-13:
OK, so the `CONTROLLER_CLIENTTIMEOUT_DFLT` for this one doesn't really make sense. (100 msec)

I guess when sending email, its receiver is probably less likely to be on the same local network, and then 100 msec is rather short.
1000 msec is much better default for this.

chaluz74, 2020-05-13:
> 
> 
> And maybe in public sites don't have the `AUTH LOGIN` part shown.
> Even though it looks encrypted, it is probably reversible, so after this, I suggest to change the password on this smtp2go account.

OK! Thanks

tonhuisman, 2023-09-02:
This seems to be resolved, so can be closed.

closed 2023-09-02 by TD-er