letscontrolit/ESPEasy #821: Framed Display: low setting too low ?

s0170071, 2018-02-04:
2.3.0 / v2.0

out of two different displays (both 3€ chinese) one is not working on display contrast setting "low".  May be a feature of the display. But with the previous "always low" bug I thought they were broken...

Possible solution: start at high contrast after boot and slowly dim down...

TD-er, 2018-02-04:
Does that display work on Medium also?
https://github.com/TD-er/ESPEasy/blob/2382cbc7ba72356b8915d7b716c5ab2280e59b0e/src/_P036_FrameOLED.ino#L218-L230

It it is working on Medium, then it is probably related to the comdetect value.
In previous builds, it was not ever set.

s0170071, 2018-02-04:
Medium is ok

TD-er, 2018-02-05:
@Grovkillen If this is a bug, then it is due to the manufacturer of the display that won't work with those settings.
The really low contrast value is perfectly fine according to the datasheet of the chip.
But is uses some extra component on the board to change the voltage on the OLED.
So I guess it i now an improvement, since with the previous build it wasn't working on that display :)

Grovkillen, 2018-02-06:
Sorry, changed it now. :wink:

TD-er, 2018-02-27:
The default value is not the low contrast value, so I consider it fixed.
If not, please re-open.

closed 2018-02-27 by TD-er