letscontrolit/ESPEasy #822: Need, if possible, incrase leght of mqtt name filed...

Lstt2005, 2018-02-05:
Dear friend! Is it possible, incrase legth of mqtt password for free service flespi.com ?In free plan very massive possibilites - https://flespi.com/pricing.
Token (it;t logon for mqtt) like -  OkEs4DE68x9J1LoMoeRUqhdk1pwkTFDvNdOHEhHYPyas9m4noJ5DO9NEm8Q4Chpf

TD-er, 2018-02-05:
How long does it have to be?

Grovkillen, 2018-02-05:
That one there is 64 bits... That might be the wanted length?

TD-er, 2018-02-05:
Somewhat related to PR #665 
Perhaps investigate what a proper length is for usernames/passwd for several controllers.

Rearranging storage is very hard, but we might be able to store some dummy string which we can either append to the user/pass when needed?

Lstt2005, 2018-02-06:
Unfortunately, the service has one significant limitation - the retain option is not supported ...

ashchurko, 2018-03-13:
Retained option for messages already integrated in flespi MQTT broker

kurczak2, 2018-10-21:
Are there any chances to increase the password field length for MQTT flespi? It would be very useful because this site does not have such limitations as, for example, CloudMqtt. With Domotocz you can connect to the MQTT flespi, but you can not send anything from espeasy :( Thanks in advance

itProfi, 2018-10-22:
Absolutely agree!!

uzi18, 2018-10-22:
Have used cloudmqtt without any problem.
how log field you need?

kurczak2, 2018-10-22:
In CloudMQTT, only 5 simultaneous connections are possible in the free plan, eg 5 Sonoffs with espeasy. There is no such limit in flespiMQTT. However, you can not enter a username longer than 25 characters to espeasy - 64 characters are required for flespi. I checked mega-20181021.

TD-er, 2018-10-22:
Maybe we can define something like this to be stored in the `CustomControllerSettings`
What sizes are needed for username and password?

itProfi, 2018-11-05:
Token (like login for mqtt) like - OkEs4DE68x9J1LoMoeRUqhdk1pwkTFDvNdOHEhHYPyas9m4noJ5DO9NEm8Q4Chpf

closed 2020-07-18 by itProfi