letscontrolit/ESPEasy #824: Thingspeak IP address outdated

s0170071, 2018-02-06:
240rc2, mega

I got a couple  of sensors sending data to thingspeak.com
All are configured as "use hostname" and host"api.thingspeak.com". In the Controllers overview some show IP address 52.1.229.129, others 52.5.13.84. 
Ping shows api.thingspeak.com @ 52.5.13.84
Restarting the esp does not change the IP, re-config does.

TD-er, 2018-02-06:
Can't you set it to DNS?
That one should do DNS lookups. (quite a recent change in ESPeasy)
There is however some caching involved, but that's on the Core library side.

I could add some flush DNS cache when the other side is unreachable.

s0170071, 2018-02-06:
Nope. The dropdown menu offers "use hostname" and "use "IP address". 
Its not a cache issue I think. Rebooting does not update the IP in the controllers overview either.

TD-er, 2018-02-06:
With "DNS" I meant the "use hostname".
Previous builds only did a lookup on the DNS and stored the IP address and kept using only the IP.
When you select "use hostname" it should do a lookup in the DNS every now and then.

So if you're using a fairly recent version of the code, it should work like that. Or I missed one in the process of fixing it.
What controller are you using to set these values? Or is it a plugin?

s0170071, 2018-02-06:
The mega version is about 3 months old. Difficult to say exacly. 20000Mega.
I'll try a more recent one then. Thanks.

TD-er, 2018-02-06:
Hmm, just looked at the code and I am not entirely sure I fixed this part.
The code looks a bit strange to me:
https://github.com/letscontrolit/ESPEasy/blob/d87cd7eafe35a24a2a3e790a5fe714d2958ddfa2/src/_C004.ino#L64-L66

Will have a look at it this evening.
And also what kind of information can be shown in the controller (and/or the log)

TD-er, 2018-02-06:
Hmm, making  connection to the server should work.
So when using DNS it should use DNS.

The other part I linked is about some HTTP Post header which apparently must include a hostname, not an IP-address.

TL;DR
It should work with recent builds.

TD-er, 2018-02-27:
I think it can be closed. If not, please re-open.

s0170071, 2018-03-09:
just tested, it got worse. If I want to set up a controller now, its trying to connect to 0.0.0.0.
It then hangs until it times out.
Setting an IP or hostname doesn't change that.

```
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat
HTTP : connecting to 0.0.0.0:80
HTTP : connection failed
HTTP : connecting to 0.0.0.0:80
Host unreachable: 0
HTTP : connection failed
Ram usage: Webserver only: 1680 including Core: 0
WD   : Uptime 1 ConnectFailures 2 FreeMem 15336

```

TD-er, 2018-03-09:
OK, will look at it this evening.

s0170071, 2018-03-09:
I looked at the code and I don't quite understand it. 

1.  Why cant there be a client.connect(host,port) where host is a string. 
2. There is no need to distinguish between IP and hostname/DNS as it is resolved by the wificlient anyway. You can throw "58.3.4.3" at it or "api.thingspeak.com". Works in both cases.

TD-er, 2018-03-09:
One thing I can come up with right now is that the WiFiclient does DNS resolves a lot more often than the one in ControllerSettings.
And these tend to hang the ESP when the DNS is slow or unable to resolve.

TD-er, 2018-03-11:
Could you please test if this now works like it should?

s0170071, 2018-03-12:
It works, but...
when you first try to set it up the web page hangs. Serial shows it trying to connect to 0.0.0.0. or 4257802018.  Web page resumes loading after it times out.
OK, 4257802018 was after I accidentally set the port to 1 instead of 80. I think I’ve seen this before...

```
 am usage: Webserver only: 0 including Core: 0
HTTP : connecting to 0.0.0.0:80
Host unreachable: 0
HTTP : connection failed
HTTP : connecting to 0.0.0.0:80
Host unreachable: 0
HTTP : connection failed
Ram usage: Webserver only: 2376 including Core: 0
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat
HTTP : connecting to :0
HTTP : connection failed
HTTP : connecting to :0
HTTP : connection failed
Ram usage: Webserver only: 1600 including Core: 0
WD   : Uptime 3 ConnectFailures 4 FreeMem 17296
WD   : Uptime 4 ConnectFailures 4 FreeMem 17296
WD   : Uptime 4 ConnectFailures 4 FreeMem 17296
WD   : Uptime 5 ConnectFailures 4 FreeMem 17128
WD   : Uptime 5 ConnectFailures 4 FreeMem 17128
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat
HTTP : connecting to api.thingspeak.com:1
Host unreachable: 4257802018
HTTP : connection failed
HTTP : connecting to api.thingspeak.com:1
Host unreachable: 4257802018
HTTP : connection failed
Ram usage: Webserver only: 2176 including Core: 0
IP blocked: 0.0.0.0 Allowed: 192.168.1.0 - 192.168.1.255
Ram usage: Webserver only: 0 including Core: 0
Ram usage: Webserver only: 0 including Core: 0
HTTP : connecting to api.thingspeak.com:1
Host unreachable: 4257802018
HTTP : connection failed
HTTP : connecting to api.thingspeak.com:1
Host unreachable: 4257802018
HTTP : connection failed

```

TD-er, 2018-03-12:
4257802018 = 253.200.231.34  (not entirely sure about the order of the bytes)
Maybe that's one of the IPs used for the service, probably based on your location.
But it's unreachable.

I think we definitely need to log proper IP-addresses and not the int representation of it.

s0170071, 2018-03-12:
api thingspeak.com resolves to 52.5.100.160 from here...

s0170071, 2018-03-12:
and there is again a 0.0.0.0 ip blocked message in between....

TD-er, 2018-03-12:
0.0.0.0 blocked is not related to this problem.
That's because of the IP filtering:
See Config => Client IP filtering => Client IP block level

s0170071, 2018-03-15:
Thingspeak is working now, however I found that the message delay option, which should delay a controller message if one has just been sent, stalls the system on every plugin read. It doesn't matter if a plugin is supposed to send data or not. 

Thingspeak is, however, demanding that requests are sent no more often than every 15 seconds. So I tried the MessageDelay (15000) . I spend the whole day figuring out why my modbus plugin was failing....
Its a new issue, isn't it ?

TD-er, 2018-03-15:
Yep, there is no reason why some obligated delay should stall the system.
If a plugin needs to do rate limits, it should store the messages, or combine them (only send latest?)
But that's something what may affect a lot of things, so definitely worth an issue.

mvdbro, 2018-03-22:
This issue seems fixed. Support for Thingspeak has never been great because we have no message queues in ESPEasy. Current design limitation, but maybe something for future releases

closed 2018-03-22 by mvdbro

s0170071, 2018-03-22:
I was planning on pimping it a bit. It it is acceptable that the plugin is re-triggered / called once per second, I could implement a buffered send that accumulates all values to be written 
a) for one second on value update (always) 
b) for 15 seconds if the previous write was just now.

a) allows multiple sensors to be accumulated and sent to thingspeak with a single request at the same interval, avoiding nan entries for unused fields.
b) is "the" annoying thingspeak limitation. a) brings some relief already...

Each delay non blocking of course, hence the periodical calls to the plugin.