letscontrolit/ESPEasy #825: feature request: Thingspeak password field as plain text

s0170071, 2018-02-06:
The Thingspeak password field is showing only dots. I know, this is a security feature. 
It does, however, make it impossible to reconstruct to which thingspeak channel the data is going to.
I would opt for making it a normal text input field as it is not really a security issue.

 Usually you copy and paste this key from the thingspeak website where it is shown in large letters. Anyone looking over your shoulder could fetch it from there...

TD-er, 2018-02-06:
We could perhaps add a "show passwd" button next to passwd text fields?

s0170071, 2018-02-06:
Why not just make it a plain text field ?
I need to see if there is a thingspeak api command to retrieve the channel ID from that key. Showing the channel ID would be even better than the api key.

TD-er, 2018-02-06:
If we make all passwd fields have a "show" option, it would be the best of both worlds I guess.
We must start to look at security also (SSL/TLS is a no-go for v2.0 by the way, will lead to crashes, I tested) and better start doing that one step at a time.

And if there is more to show, like a channel ID, that would also be possible.

What controller are we talking about?

s0170071, 2018-02-06:
Thingspeak. e.g. https://thingspeak.com/channels/84008

TD-er, 2018-02-06:
I meant the controller in ESPeasy

TD-er, 2018-02-06:
Where do you set the settings for Thingspeak in the ESPeasy?
How does ESPeasy know where to send the data?

s0170071, 2018-02-06:
1. set up a controller. e.g. http://192.168.1.15/controllers. Press edit, select thingspeak.
2. the data comes from the devices. Select http://192.168.1.15/devices?index=1&page=1
3. activate controllers (Data acquisition-> send to controller), adjust IDX to 1.

sassod, 2018-02-11:
> I looked into Thingspeak, to get an idea of what information is relevant.
> But you cannot create a free account without a Mathworks account.
> And since I do not have a Matlab license, I cannot create a Thingspeak
> account.
>
> Can you show some screenshots (with masking parts of personal info) on what
> kind of information should be relevant?
> I find it a bit hard to believe the password should be the only key
> information on seeing the active channel.

There is no need as far as i know to have a mathlab license to use
thingspeak. I don't have a license but i have a lot of sensors connected to
thingspeak.
The password is not an actual password it is a api key generated for
write/read access to a particular chanel the user generates and the idx is
the number (1-8) of the field that the value is written to.

TD-er, 2018-02-11:
Hmm, looks like I was not reading good enough indeed.
Still waiting on my activation email. Then I will have a look on how to use it.

TD-er, 2018-02-11:
Hmm it looks like the Controller interface of ThingSpeak should have a name field and an API key field.
If they are used like that, they should be called the same in the user interface of ESPeasy.

TD-er, 2018-02-11:
Can you have a look at the commit + PR I just made?

sassod, 2018-02-12:
As I understand thingspeak the needed fields to push values is only WRITE
API KEY (controller tab) and CHANEL ID (IDX device tab)  to whitch api key
is bonded.
The name is not important as that can be anything as long as the idx is
correct.

The pass field should be Write API key ( not read api key!! from thingspeak
chanell)

The IP/hostname and the port are always the same as described and could be
removed from the web gui and hardcoded to the plugin as there is no private
server available for download or setup.

Addresses and Locations

Regular URL: [cce]http://api.thingspeak.com[/cce]
Secure URL: [cce]https://api.thingspeak.com[/cce]
IP Address: [cce]http://184.106.153.149[/cce]
Cross-domain XML: [cce]http://api.thingspeak.com/crossdomain.xml[/cce]ž

TD-er, 2018-02-12:
The name was just an indication to identify the used API-key, just like the original issue requested. (to ease identification)

The label could indeed be changed. I just took the same labels as displayed in the Thingspeak account.

I don't know where the cross-domain XML is used.
Is there no local install possible for thingspeak? Or some paid version that allows for other domainnames?

sassod, 2018-02-12:
Hm never thought that a local server is possible as I thought the code is
not available... I never looked around to find out but as I can see now it
is open source and will test this afternoon on a vps to check if it is
working.....

sassod, 2018-02-13:
ok I have been bashing my head to install thingspeak on a vps for the past
2 days and I just cant install it ( ruby on rails is too much for me to
configure at the moment as I don't know it even a little bit ).

maybe someone else with more knowledge can test this if it is
working....sorry guys

closed 2018-02-27 by TD-er