letscontrolit/ESPEasy #827: Issue with MQTT not using proper case for %sysname% variable.

mattlward, 2018-02-06:
### Steps to reproduce
Set up rules with MQTT publishes.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

Yes

### Expected behavior
on System#Boot do
 Publish %sysname%/status,Booted
 Publish %sysname%/IP,%ip%
 Publish ESP_Work/status,Booted
 Publish ESP_Work/status,WTF
 Publish %sysname%/status,wtf 

I would expect the following:
ESP_Work/status   Booted
ESP_Work/IP xxx.xxx.xxx.xxx
ESP_Work/status   Booted
ESP_Work/status  WTF
ESP_Work/status  wtf

### Actual behavior
Results in MQTTfx:
esp_work/status   booted
esp_work/ip   xxx.xxx.xxx.xxx
esp_work/status   booted
esp_work/status   wtf
esp_work/status   wtf

Cases are ignored in rules.  When defined in the devices tab the work correctly for my temperature I use the following for my SHT30 module:
name = SHT30
variable = Temperature 
The result in MQTTfx is:
ESP_Work/SHT30/Temperature  72.1

### System configuration

Hardware:
Wemos D1 V3.0.0
Wemos SHT30 shield

Software or git version:
v2.0-20180206 Dev build
Has been a problem for all versions I have tested.

TD-er, 2018-02-06:
I just tested with [MQTTBox](https://chrome.google.com/webstore/detail/mqttbox/kaajoficamnjijhkeomgfljpicifbkaf/related?utm_source=chrome-ntp-launcher)

And also tried to log the strings just to see where the case is lost:

```
EVENT: Rules#Timer=1
ACT  : Publish ESP_EasyM/status,Booted
MQTT : Publish ESP_EasyM/status,Booted
ACT  : Publish ESP_EasyM/IP,192.168.1.75
MQTT : Publish ESP_EasyM/IP,192.168.1.75
ACT  : Publish ESP_Work/status,Booted
MQTT : Publish ESP_Work/status,Booted
ACT  : Publish ESP_Work/status,WTF
MQTT : Publish ESP_Work/status,WTF
ACT  : Publish ESP_EasyM/status,wtf
MQTT : Publish ESP_EasyM/status,wtf
ACT  : timerSet,1,10
```

Using these rules:
```
on System#Boot do
  timerSet,1,10
endon

On Rules#Timer=1 do
  Publish %sysname%/status,Booted
  Publish %sysname%/IP,%ip%
  Publish ESP_Work/status,Booted
  Publish ESP_Work/status,WTF
  Publish %sysname%/status,wtf
  timerSet,1,10
endon
```

As you can see, the capitalization is still there:
[image]

What MQTT controller do you use?
I tested with Domoticz MQTT

Grovkillen, 2018-02-06:
This used to be a problem way back. Working for me using openHAB (but as a generic mqtt)

mattlward, 2018-02-07:
I am using the openHAB MQTT controller.  This really makes it hard to code for expected results.

mattlward, 2018-02-15:
Ok, so here are my rules for publishing on boot:
On System#Boot do    //When the ESP boots, do
 Publish %sysname%/IP,%ip%
 Publish %sysname%/status,Booted
 Publish ESP_MBR/status,Connected
  timerSet,1,30      //Set Timer 1 for the next event in 300 seconds
endon

I used MQTTBox for the test since you chose that program and it is not respecting case!
[image]

TD-er, 2018-02-16:
I think I see what you mean...
The topics you get to see, seem to be all lowercase.

I just finished some other MQTT-related PR, so I had it all setup with a topic with some part uppercase ("LWT")
See here both subscribed to the same broker, only one subscribed to the proper topic and one to the same topic in lower case:
[image]

As you can see, here the one with the correct topic (case sensitive) is showing the LWT messages and the other is not.
So ESPeasy is sending it with the correct case and it looks like your broker is not case sensitive. (perhaps a setting somewhere?)

I only used MQTTbox, since that was the second program I tried and it worked.
If there is a better one, please feel free to suggest one.

mattlward, 2018-02-16:
My point is that they are not all lower case!  Something is not coming out of the ESP properly.  My broker, mosquito on hassio sees them both.  But, I am having to build mqtt sensors in all lower case for some things and mixed case for others. 

So, it appears that if the data comes from one of the tasks or devices that case is properly presented, but if it is generated by a publish from a rule the case is lower...

[image]

I use MQTTfx and love it, very fast and very clean.

TD-er, 2018-02-16:
I have looked into the rules source code and one thing that struck me as odd, is that the rules use 2 strings intermittently. One in original case and one small case.
This is probably being done to ease the string matching of keywords. 
But I guess there is some mix-up of strings somewhere.
Maybe we can use 1 string and use a case insensitive string match to find keywords.

mattlward, 2018-02-16:
@TD-er it sounds like you may be on to something.  There is definitely something wrong with case coming out of the rules... way above me, but writing HA sensors for every case is a pain.

mattlward, 2018-02-16:
Do I need to put it a fresh issue, now that I have found a little more info?

Thanks

TD-er, 2018-02-16:
Nope, this one has all the information and I think the only thing changed
is that I now kind of understand the problems you're facing ;)

---
Verzonden vanaf laptop

mattlward, 2018-02-16:
Great!  I am not good at explaining the issue yet, still learning the proper terms and ESPEasy.  Thanks for the help!

TD-er, 2018-02-17:
Issue #847 took a lot more time, so could not look into the rules.
Will try tomorrow evening.

mattlward, 2018-02-17:
No problem, can't test until Sunday eve.

mattlward, 2018-02-20:
This is still an issue after the latest nightly build, I noticed many MQTT and file fixes and thought it might have caught it.

TD-er, 2018-02-20:
Nope, did not fix it yet.
It requires quite some changes in the rules and since I was doing a lot for the MQTT, I though it would be better to stay at one subject and fix it properly and then change to the Rules.
I did already some minor preparations, but the rules part is quite new to me, which will require some reading for me.

The next two evenings are also not entirely dedicated to me programming ESPeasy, so a few more days I hope.

mattlward, 2018-02-21:
Did you see my comment that these are working correctly in the build that you linked to me in the thread regarding web pages not loading?

TD-er, 2018-02-21:
No, did not see that.
That build was based on the mega branch, which means there still is a bug in V2.0

TD-er, 2018-02-27:
Is this still an issue with the latest v2.0 builds?

mattlward, 2018-02-28:
Let me test, what version do you want me to test?

mattlward, 2018-02-28:
It does not appear to be fixed in v2.0-20180228

This rule:
on sniff#cmdMQTT=1 do
  Publish %sysname%/IP,%ip%
endon

Returns esp_br/ip/xxx.xxx.xxx.xxx

mattlward, 2018-02-28:
The same rule with mega-20180228 produces the following:
ESP_Br/IP/xxx.xxx.xxx.xxx

TD-er, 2018-02-28:
Ok, I will look at it this afternoon and compare version files.
Thanks for testing

TD-er, 2018-03-03:
OK, that was one deep dive into the code.

Is it possible you were using one of these plugins:

- P054 DMX512
- P057 H16K33_LED
- P065 DRF0299_MP3
- P067 HX711_Load_Cell
- P070 NeoPixel_Clock

These could change the case in a string which was being used in the processing of rules.
Exactly that case altering behavior was changed in Mega branch for all these plugins and not in v2.0 branch.

mattlward, 2018-03-03:
Nope, none of them.

TD-er, 2018-03-03:
I did change some other case related things also, could you test with the next build?

mattlward, 2018-03-03:
Loading now!

mattlward, 2018-03-03:
Not a good sign, flashed 0228 and it lost its mind... corrupted flash.  Rebuilding now.

mattlward, 2018-03-03:
Ok, got it loaded and the case in 0228 is still all lower.  Of course mega latest is ok.

TD-er, 2018-03-03:
Why test with build of feb 28?
There is a build of this morning: https://github.com/letscontrolit/ESPEasy/releases/tag/v2.0-20180303

mattlward, 2018-03-03:
Will test it today... In the middle of flashing a new S31.

mattlward, 2018-03-03:
Getting closer!  The following rule is in play:

On System#Boot do    //When the ESP boots, do
 Publish %sysname%/IP,%ip%
 Publish %sysname%/status,Booted
  timerSet,1,30      //Set Timer 1 for the next event in 300 seconds
endon

The first publish is right!  The second publish pushes out esp_spare/status/booted, all case ignored on publish.

TD-er, 2018-03-03:
And if you repeat it, thus in the timer1 routine?
Is only the first publish with correct case?

mattlward, 2018-03-03:
I will try that... That is running the build that you sent.

mattlward, 2018-03-04:
So, I made some changes to the rules to test the boot values versus the after boot (timer) values.
What I found is that latter runs now honor case and that the publishes that should happen at boot do not happen at all.

On System#Boot do    //When the ESP boots, do
 Publish %sysname%/IPb,%ip%
 Publish %sysname%/Statusb,Booted
  timerSet,1,30      //Set Timer 1 for the next event in 300 seconds
endon

On Rules#Timer=1 do  //When Timer1 expires, do
 Publish %sysname%/IPt,%ip%
 Publish %sysname%/Statust,Booted
  timerSet,1,30       //Resets the Timer 1 for another 300 seconds
endon

mattlward, 2018-03-04:
As you can see they are all sub(t) from the repeating timer, but case is great!  The first rule does not seem to process.
[image]

TD-er, 2018-03-04:
Could you also have a look at the logs?
There should be something in there stating: "PLUGIN_WRITE altered the string:"

mattlward, 2018-03-04:
Ok, syslogging on and I see the first set of publishes happening before wifi connects.  Going to throw a timer in and will report back...

Making this rule change, will report later... gotta go.

mattlward, 2018-03-04:
I will also find the thing that scrapes all of the screens.  This does have 2 MQTT import tasks running.

mattlward, 2018-03-04:
I did not see "PLUGIN_WRITE altered the string:" in the log or in syslog.  

I have attached screenshots, rules files and full syslog of the first minute of a reboot.

It appears that the first part of rules1.txt is trying to process but MQTT connection is not up yet.  

[image]
[image]
[image]
[rules1.txt](https://github.com/letscontrolit/ESPEasy/files/1778201/rules1.txt)
[rules2.txt](https://github.com/letscontrolit/ESPEasy/files/1778202/rules2.txt)
[rules3.txt](https://github.com/letscontrolit/ESPEasy/files/1778203/rules3.txt)
[image]
[image]
[image]
[image]
[image]

mattlward, 2018-03-04:
I figured out how to do the delay properly!  I am now getting the bootup MQTT notification and the first timer expiration message thru to the broker.  I think the case problem is solved, but should it try to publish before mqtt is connected?  Should that first delay be required?

[image]

New rules1:
On System#Boot do    //When the ESP boots, do
 timerSet,1,15
endon

On Rules#Timer=1 do  //When Timer1 expires, do
 Publish %sysname%/IPb,%ip%
 Publish %sysname%/Statusb,Booted
  timerSet,1,60      //Set Timer 1 for the next event in 300 seconds
endon

On Rules#Timer=1 do  //When Timer1 expires, do
 Publish %sysname%/IPt,%ip%
 Publish %sysname%/Statust,Booted
  timerSet,1,60       //Resets the Timer 1 for another 300 seconds
endon

mattlward, 2018-03-04:
Tried to make a production boot up rule set... It is working and topics make more sense now.

On System#Boot do    //When the ESP boots, do
 timerSet,1,15
endon

On Rules#Timer=1 do  //When Timer1 expires, do
 Publish %sysname%/Status/,%ip%
 Publish %sysname%/Status,Booted
  timerSet,1,60      //Set Timer 1 for the next event in 300 seconds
endon

On Rules#Timer=1 do  //When Timer1 expires, do
 Publish %sysname%/Status/,%ip%
 Publish %sysname%/Status,Connected
  timerSet,1,300       //Resets the Timer 1 for another 300 seconds
endon

[image]

mattlward, 2018-03-04:
New production rule 1 for me:

On System#Boot do    //When the ESP boots, do
 timerSet,1,15
endon

On Rules#Timer=1 do  //When Timer1 expires, do
 Publish %sysname%/Status/IP,%ip%
 Publish %sysname%/Status/Unit,Booted
  timerSet,1,60      //Set Timer 1 for the next event in 300 seconds
endon

On Rules#Timer=1 do  //When Timer1 expires, do
 Publish %sysname%/Status/IP,%ip%
 Publish %sysname%/Status/Unit,Connected
  timerSet,1,300       //Resets the Timer 1 for another 300 seconds
endon

[image]

mattlward, 2018-03-04:
Yup, second MQTT import was affected... not able to load the last 2 values.

TD-er, 2018-03-04:
Currently there can only be one MQTT import active.
I was looking into allowing more, but that looks to be quite some code change.

mattlward, 2018-03-05:
I am going to back down, really need the 2 mqtt importers.  This test unit is a central weather display unit.  I need 6 import tasks, 7 and 8 are for control from HA.  I will look into pushing the values out via http, looks like it may be possible but it is not a real fix for me.

TD-er, 2018-03-05:
Currently there is a hard limit of 4 values per task.
Not sure what will break when increasing that limit.
Probably storage layout

mattlward, 2018-03-05:
I guess I am going to figure out how to get the variables in another way... going to be much less convenient than using MQTT for sure.

closed 2018-03-25 by Grovkillen