letscontrolit/ESPEasy #828: Publish to MQTT from rule stopped working

arduinopjotr, 2018-02-06:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

With firmware ESPEasy_v2.0-20180124 or later (testen till ESPEasy_v2.0-20180206) i have an issue with
sending (publish) MQTT message within rule. for example:

on Pushbutton#SW04 do                       // trigger wanneer switch veranderd toggleswitch
  if [RelayStatus#SW05]=1 do
    gpio,5,0                                           // Relais uit
    Publish /Bewatering/Power,Off
//    TimerSet 1,60

When i load firmware ESPEasy_mega-20180114 or earlier everything works fine.
 

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
Message published with MQTT within rule

### Actual behavior
Tell us what happens instead?
nothing

### System configuration

Hardware:
Wemos D1

Software or git version:
mega-20180114 development 
ESP_Easy_mega-20180114_dev_ESP8266_4096
[rules1.txt](https://github.com/letscontrolit/ESPEasy/files/1700111/rules1.txt)

TD-er, 2018-02-06:
The only notable changes between 20180114 and 20180116 were:

1. My daughter became 4 yrs old
2. Some changes in time related functions (Auto DST and move of time functions)

Both are somewhat time related, but I guess I'll have to look into the source code, since I will not try to wake my daughter and ask her about watering the plants :)
So let's hope the problem is in the code, or else you have to wait ;)

TD-er, 2018-02-06:
Hmm, maybe @Grovkillen can also have a look at your rules first, before I dive into C++ code without knowing what to look for.

uzi18, 2018-02-06:
Please check if this command work in advenced options tab:
Publish /Bewatering/Power,Off

Grovkillen, 2018-02-06:
Publish /Bewatering/Power,Off

Works for me.

TD-er, 2018-02-14:
So it must probably have something to do with the time functions I moved?

uzi18, 2018-02-14:
@TD-er no, MQTT.loop() was not launched before sleep.

uzi18, 2018-02-14:
Sorry not here :D

diehardsk, 2018-02-22:
I met the same problem with **2.0** firmware (ESP_Easy_v2.0-20180221_test_ESP8266_4096.bin)
But mega firmware is OK for me. Please try. (ESP_Easy_mega-20180221_test_ESP8266_4096.bin)

Controller: OpenHAB MQTT
MQTT Broker: Mosquitto
Rules command: Publish /%sysname%/LEDswitch/state,1

uzi18, 2018-02-23:
On the other hand mqtt.loop() will not be executed if deepsleep is enabled.

TD-er, 2018-02-23:
@uzi18 That's an interesting remark you just made.
Will have to think about it, how the new idea of timers should be implemented.

TD-er, 2018-03-02:
Maybe somehow related to #827 ?
OpenHAB MQTT + rules + works in Mega, not in v2.0

TD-er, 2018-03-23:
Is this still an issue?

arduinopjotr, 2018-03-23:
Hi Gijs,

 

I shall try it with the latest build

 

Regards,

 

Peter van der Veer

Grovkillen, 2018-03-25:
A report please? :)

mgoepel, 2018-03-31:
ok, further to this
I have a mosquitto broker running on Raspbian Stretch - all latest versions
My IP address publishing rule worked on the previous versions of ESPEASY - as above, and then stopped working.
I just loaded up mega-20180331 DEV versions on many different ESP's and the rules publishing to MQTT appears to be working again!!!

Thank you.

Out of interest, here is my rule set

On System#Boot do    //When the ESP boots, do
  Publish Instrument/%sysname%/SY/IP,%ip%
  timerSet,1,20      //Set Timer 1 for the next event in 20 seconds
endon
On Rules#Timer=1 do  //When Timer1 expires, do
  Publish Instrument/%sysname%/SY/IP,%ip%
  timerSet,1,60       //Resets the Timer 1 for another 60 seconds
endon

closed 2018-04-06 by Grovkillen

TD-er, 2018-05-05:
Could you please also test with the last release I made last night, not the nightly build.
It is in the releases section.
There has been an issue with the compilation of builds lately

Oxyandy, 2018-05-05:
@ScanxTaz  [Download from Here](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180505_TD-er)

ScanxTaz, 2018-05-08:
Thanks, will give a try this week and let you know my findings

ScanxTaz, 2018-05-14:
testing... Having some difficulties to find which image I should choose from the zip you sent me :)

TD-er, 2018-05-14:
You should be able to use the nightly builds again. (not todays build, which is deleted because of a bug)
For example 20180513.
And then pick the version based on the processor you have (ESP8266 or 8285) and the amount of flash you have.
You could see this information in the sysinfo page on the current version you run.

ScanxTaz, 2018-05-15:
Yo, thanks for your support. I'm currently trying to reset my sonoff, but it seems i've issue with the flashing, I suspect the DOUT symptom.

I'm using my mife's laptop on #fuckdows so will give a try tonight with my ubuntu, will be better I think... Oh... It's maybe me to you now :) #nobodyIsPerfect :)

Keep you posted

ScanxTaz, 2018-05-26:
OK, you can close the issue, working on my side...

It took me a Loooooonnnng time to realize that the config is kept when I'm upgrading :) lol
Thought I would have to reconfigure everything...

Wonderful job guys, thanks.... I'll continue to follow and post logs in case of issue.