letscontrolit/ESPEasy #836: Feature request color display

s0170071, 2018-02-09:
I have a couple of ST7735 displays 
https://www.arduino.cc/en/Guide/TFT
They are cheap and SPI. Thought they may be added to the framed display plugin.

TD-er, 2018-02-09:
Hmm, SPI and ESPeasy is quite a new combination I guess.
Never seen any plugin using SPI.

I also have a SPI epaper display which still has to be supported.

Thing with color displays is that the required memory for the frame buffer will increase a lot compared to 1 bit per pixel displays.

s0170071, 2018-02-09:
I looked at the library. The interface is encapsulated nicely. Just in case I happen to get bored a lot I'll do it...

TD-er, 2018-02-09:
Please also have a look at these repo's: https://github.com/squix78?tab=repositories
This is from the author of the OLED library.
It looks like he is making some generic display library for all kinds of displays to connect to the ESP8266.

He abandoned the OLED-SSD1306 library (no PRs merged for over a year) and I took all suggestions and PRs from that github and collected them in the update of the library I created a few days ago.

realwax, 2020-02-15:
+1 on this. Would be a charm having support for ST7735S out of the box. instead of being bound to the smaller 0.6 display. Thank you!

tonhuisman, 2021-08-15:
@s0170071 @realwax @drone777 If you are interested, here are .bin files with support for these displays:

[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6988679/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6988680/ESPEasy_display_ESP8266_4M1M.zip)

(ST7735: 128x128, 128x160, 80x128, ST7789 240x320, 240x240, 240x280, all connected via SPI)

realwax, 2021-08-16:
@tonhuisman Thank you!

tonhuisman, 2021-08-16:
Please report your findings, good or bad, there will soon be updates with new features and improvements 😃

tonhuisman, 2021-08-16:
Did some bugfixing and added a few features, re-organized device settings somewhat.
Updated builds are available, and also documentation:

[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6993665/ESPEasy_display_ESP8266_4M1M.zip)
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/6993668/ESPEasy_display_ESP32_4M316k.zip)

[Display - ST7735_ST7789 TFT — ESP Easy 2.1-beta1 documentation.pdf](https://github.com/letscontrolit/ESPEasy/files/6993669/Display.-.ST7735_ST7789.TFT.ESP.Easy.2.1-beta1.documentation.pdf)

realwax, 2021-08-18:
@tonhuisman Thank you for your efforts! I tested today with a Wemos D1 mini and [Wemos TFT Shield 1.4"](https://www.wemos.cc/en/latest/d1_mini_shield/tft_1_4.html) Everthing is working like a charm. Connected it via MQTT to iobroker. Colors, textspacing, textpos, drawing rectangles, ... all working! The only strange thing I encountered was when sending tftcmd,off it clears screen and bg color it goes white and not off. I cannot dim it, too. But I think it's related to the TFT shield as it states in the link TFT_LED NC - which means not connected i guess - if TFT_LED is meant to be the backlight. I guess I need a different display to enable that feature. What do you think? And again - thank you for integrating it! :)

tonhuisman, 2021-08-18:
Great that it's working (mostly) as expected 👍 
Is the backlight dimming working at all for your board, so if you set it to 30% the display gets dimmed, compared to when set to 100%?
The NC most likely means it is not connected, indeed, so I don't think the backlight dimming can work without adding extra wiring.

I have been working on the code for 2 more days already, and have done a lot of improvements, especially to the `clear` command 😉 I'll publish an updated build probably late(r) today (CEST) or tomorrow.

I own a few of these displays, 1x ST7735 (128x160) and 1x ST7789 (240x240), and where the ST7789 doesn't have a CS connection, the display and the BLK connection is working like a charm, but the ST7735, having both CS and BLK connections, the backlight is not working at all when anything is connected to that pin (and display contrast/readability is really poor because of that). I suspect it has to do with the 'Chinese quality' of these devices, as I ordered them straight from Aliexpress 😅

realwax, 2021-08-18:
As you said. Back light is not working at all. Change affects nothing. I connected the TFT_CS to D4. TFT_LED which I thought should be the back light to D0 and configured it. But on the shield as it says NC there is a need to solder or connect. The schematic should show it but I not used to understand what they mean here. https://www.wemos.cc/en/latest/_static/files/sch_tft1.4_v1.0.0.pdf
There is a bottom solder point https://www.wemos.cc/en/latest/_static/d1_shields/tft_1.44_v1.0.0_2_16x16.jpg where i guess it needs to be connected. The middle point to one of the outers https://www.wemos.cc/en/latest/_static/d1_shields/tft_1.44_v1.0.0_2_16x16.jpg

But as you said - poor build ;)

tonhuisman, 2021-08-18:
Looking closely at that picture, it seems that `TFT_CS` = `D4`, `TFT_DC` = `D3` and `TFT_RST` = `RST` (hardware reset pin from ESP), so `TFT_LED` can't be soldered to `D3` or `D4`, remaining only `D0` or `D8` for use (without major surgery).
If you connect that center island to either `D0` or `D8` islands next to it, and configure in ESPeasy, it will probably work.

realwax, 2021-08-18:
I soldered it to D8 as I messed up D0. There seems to a kind of transparent coating on the board. But it does not make any difference. Maybe my solder wasn't working or this sneaky little bas... doesn't support it - that's why NC. Anyhow thanks for pointing me towards the right direction. I don't want to spoil this thread here with a hardware issue. I posted this thread to a YT video with an immediate reaction welcoming your implementation. So maybe more testers ;) Have a nice day :)

TD-er, 2021-08-18:
D8 is pulled to GND (or else the ESP will not boot)

See [Best pins to use on ESP8266](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)

realwax, 2021-08-18:
> D8 is pulled to GND (or else the ESP will not boot)
> 
> See [Best pins to use on ESP8266](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)

@TD-er Thank you.

tonhuisman, 2021-08-19:
I've added a few features, fixed a handful of bugs and did some other improvements, so here are new builds for testing.

[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7017452/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7017453/ESPEasy_display_ESP8266_4M1M.zip)
(Documentation will be updated later, but the previous docs are still mostly correct)

Latest feature is that there is inline processing of subcommands added to the default template, by using `{:txs,4:}` for changing the font scaling (executed before displaying the line, so _no_ multiple font scaling per line), and the likes, resulting in this kind of display on my test-unit (ST7789 240x240px display):

[image]

tonhuisman, 2021-08-20:
The previous sample was a bit 'single-color', so I upgraded it to show some (maybe a little chauvinistic 🎏 😜) colors:

[image]

I've also changed the template delimiters from `{:` and `:}` to `~` and `~`, that's why the second line shows a few `~` characters, just to test that it still is working as intended 😄 
('Sun' means %sunrise% in this context, and the background-blue is still the same, but the image is taken from a somewhat different angle, demonstrating the 'quality' / viewing angle of the display, and evening vs day lighting conditions 🤔)

realwax, 2021-08-20:
> The previous sample was a bit 'single-color', so I upgraded it to show some (maybe a little chauvinistic 🎏 😜) colors:
> 
> [image]
> 
> I've also changed the template delimiters from `{:` and `:}` to `~` and `~`, that's why the second line shows a few `~` characters, just to test that `~txs,2~~ %ip% ~` still is working as intended 😄
> ('Sun' means %sunrise% in this context, and the background-blue (navy) is still the same, but the image is taken from a somewhat different angle, demonstrating the 'quality' / viewing angle of the display, and evening vs day lighting conditions 🤔)
Looking good!

I ordered myself a 2" ST7789  240x320 and will reflash on Monday and test. :)

realwax, 2021-08-24:
Hey @tonhuisman,
I got my 2" display but I got issues. It's a[https://www.waveshare.com/wiki/2inch_LCD_Module]( https://www.waveshare.com/wiki/2inch_LCD_Module ). It's connected correctly but it stays dark when I choose the correct ST7789 240x320. For testing I switched to 240x240 and 240x280. Both work but distorted. I attach a picture. But as said when choosing the right option - it's not possible to display anything. Thank you for your help/fix.

Cheers
Wolfgang
280 mode
[image]
240 mode 
[image]

320 mode :(

tonhuisman, 2021-08-24:
I've now changed initialization to also set the resolution for 240x320, as was already done for 240x240 and 240x280, maybe that improves stuff.
Also fixed a color-parsing bug when using `white`.

Please try these new downloads:

[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7041850/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7041851/ESPEasy_display_ESP8266_4M1M.zip)

realwax, 2021-08-24:
@tonhuisman Thank you very much! That was the trick. Display is now working!

HV-NL, 2021-08-24:
I do own a LilyGo TTGO-T (ESP32 with built-in 240x135 1.14" TFT screen, driven by a ST7789.
(http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1126&FId=t3:50033:3)
However 240x135 is currently not in the display list.....     Is it possible to also add this resolution?

tonhuisman, 2021-08-24:
> However 240x135 is currently not in the display list...

I'll add that resolution, those LilyGo ESP's are quite popular, it seems.

tonhuisman, 2021-08-25:
> I do own a LilyGo TTGO-T (ESP32 with built-in 240x135 1.14" TFT screen, driven by a ST7789.
> (http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1126&FId=t3:50033:3)
> However 240x135 is currently not in the display list..... Is it possible to also add this resolution?

I've added that resolution, @HV-NL please test and report your findings here.
Also investigated if other resolutions are used, but I didn't find any that aren't supported yet.

New downloads:

[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7045969/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7045972/ESPEasy_display_ESP8266_4M1M.zip)

HV-NL, 2021-08-25:
I tried to test, but no luck!
When configuring TTGO-T I cannot select the proper SPI settings on the hardware page:
```
MISO = N.C.
MOSI = 19
CLK  = 18
```

tonhuisman, 2021-08-25:
Ah, it's currently not possible in ESPEasy to select different from the H-SPI/V-SPI settings.
Have to investigate how that can be solved though, as this is hidden somewhere deep down in the dungeons of the (SPI) code 🤔

tonhuisman, 2021-08-25:
> I cannot select the proper SPI settings on the hardware page:

Have you tried to confirm it really didn't work with the VSPI option on the Hardware page? It _may_ be a case of poor labeling, as happens rather often with China-based hardware 😞

drone777, 2021-08-25:
This config works in arduino sketch:

| Name | Pin |
| ---- | :----: |
| TFT_MISO | 	N/A |
| TFT_MOSI | 	19 |
| TFT_SCLK	| 18 |
| TFT_CS	| 5 |
| TFT_DC	| 23 |
| TFT_RST	| 26 |
| TFT_BL	| 27 |

On ESPEasy test firmware backlight works.

tonhuisman, 2021-08-25:
Hmm, also looked at some other LilyGo modules, like the ones that come with a camera, and they use even _different_ SPI pins, and I've also seen different pin assignments for different hardware revisions of some of these modules, so it is quite a mess there. 

@TD-er Should we go for a 3rd SPI option for ESP32 where these 3 SPI pins can be 'freely' configured, similarly to the I2C pins? Re-wiring these modules doesn't seem like a feasible option, AFAICS.

HV-NL, 2021-08-25:
> 
> 
> > I cannot select the proper SPI settings on the hardware page:
> 
> Have you tried to confirm it really didn't work with the VSPI option on the Hardware page? It _may_ be a case of poor labeling, as happens rather often with China-based hardware 😞

Yes, I tried. Only the backlight is working, the display stays totally black. No luck.

TD-er, 2021-08-25:
Yep, that is probably a good option to make it configurable.
ESP32 can map almost any pin to almost any function.

We should define a SPI bus in the hardware tab, like we do with I2C. (I guess that's what you meant too?)
That does make more sense I guess, as we "only" have 2 SPI busses on ESP32.

tonhuisman, 2021-08-31:
I've been bug fixing (and waiting a short while for such module 😄), so now the rotation related issues have been resolved. Tested using 135x240, 240x240 and 240x320 ST7789 displays.

Here is the ESP32 build that includes the User-defined SPI GPIO settings:

[ESP_Easy_mega_20210831_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7086309/ESP_Easy_mega_20210831_display_ESP32_4M316k.zip)

HV-NL, 2021-09-04:
I did some tests, but the results are not yet as I expected. Rotation 'normal' and rotation '+180°' look fine, but rotation '+90°' and rotation '+270°' give unexpected results. (I use the default 'truncate exceeding message' during my tests)

I did my tests with some long and some shorter lines:
[image]

This is initially shown on the 135x240 display (looks fine):
[image]

But when I start with rotation 90, then (after a reset, so starting with an empty screen) all lines are partially truncated, e.g.
"line 04 line 04 line 0"
i.s.o.
"line 04 line 04 line 04 line 04"

[image]

Next test: no reset, but use a command to rotate. I start with rotation 'normal' or '90°', so the screen is filled normally. When I next submit the command "st7789,rot,1", followed by "taskrun,display_240_135"  I see text in 2 directions (because the lines are truncated I partially still see the original contents)

[image]

tonhuisman, 2021-09-04:
Hm, that sounds exactly like the behavior *before* I fixed those rotation bugs, so either I uploaded the wrong build zip, or your flash update failed 🤔 
I've fixed more bugs today, so a new build is available. Because of the specific needs of this hardware I'm including the User-defined SPI settings feature (ESP32 only) here:

[ESP_Easy_mega_20210904_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7109807/ESP_Easy_mega_20210904_display_ESP32_4M316k.zip)

And the regular Github actions build for ESP8266:

[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7109812/ESPEasy_display_ESP8266_4M1M.zip)

I've flashed it on my test-unit, and the initial display, on +90 degrees, looks like this:
(initial also means the IP address is still `0.0.0.0`, and Sunrise/sunset 0:00)

[image]
(I really don't like that the backlight pin is connected to GPIO 4, same as the on-board WiFi led (blue light on left bottom) 😞)

This is my Content: (nb: Text print Mode is set to `Truncate exceeding message`, and background color is `navy`)

[image]

HV-NL, 2021-09-04:
I was testing with version Aug 31 2021 21:12:22

You are initialising the display yourself, due to the first line `"~txc,white,blue~~txs,3~ESPEASY"`. In my tests I did not instruct the display to use any other color or size than default.

I just did a quick test with the latest version. Initially I got exactly the same (truncated) lines as before (using my test configuration as described previously). Then I entered your text in line 1, 2 and 3. That was BINGO: now the lines were no longer truncated.

So I tried a mix of my lines and your lines (i.e. re-initialise at some moment (line 2), and after that line no trucation occurs any more)

This is my new test configuration:
```
%ip% line 01 line 01 line 01 line 01
~txc,blue,white~~txs,2~ST7789 ST7789 ST7789
~txc,white,navy~~txs,1~%ip% line 03 line 03 line 03
line 04 line 04 line 04 line 04 line 04
abcdfefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
line 06 line 06 line 06 line 06 line 06  
line 07 line 07 line 07 line 07 line 07 
```
Please look to the first line: that line is still trucated. Line 2 and all others are OK (i.e. not truncated), seemingly due to the re-initialisation that has occurred in line 2 
`"~txc,blue,white~~txs,2~ST7789 ST7789 ST7789"
`![DSCN4107](https://user-images.githubusercontent.com/33630544/132099350-17c6633a-1954-4769-8dc0-a7bf6e5e415e.jpg)

tonhuisman, 2021-09-05:
It took a bit of time, but I found the bug, and fixed it. Thanks for pointing it out! As usual, any assumption (I) made is wrong 😄
(The text size wasn't recalculated when changing the rotation, but the `txs,2` command does that in the example. Using `txs,1` in line 1 would have also 'fixed' it.)

A new build is available (ESP32 including the User-defined SPI pins feature):
[ESP_Easy_mega_20210905_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7111683/ESP_Easy_mega_20210905_display_ESP32_4M316k.zip)
And for ESP8266 the regular Github actions build:
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7111705/ESPEasy_display_ESP8266_4M1M.zip)

TD-er, 2021-09-05:
Just curious, how difficult is it to (optionally) make large fonts with high resolution?
It could also be a subset of a full character set if it makes the fonts unusable large.

I just bought this bench multimeter and I like the way how it represents values and settings:
[image]
So maybe we can make menu page helpers later to allow for similar instrumental setups.

Just as an inspiration, not to copy/clone it.

tonhuisman, 2021-09-05:
Yes, that's looking quite nice. (and inspiring)
The standard font 'scaling' is a multiplication factor, so currently larger font sizes are quite 'blocky', no interpolation at all is done. The extra fonts (to be enabled in a custom build) add quite some size to the .bin file, but could also be workable on a MAX build, I think.
The font conversion site [here](https://rop.nl/truetype2gfx/) can take any true-type font and convert it, but most usable are mono-spaced fonts. I'll convert & add a few more, (somewhat) larger fonts, just for fun.

TD-er, 2021-09-05:
Can you also use (small) bitmaps?
Just trying to gather some ideas in my head.

Just thinking out loud, I am thinking of some state handler concept to allow automations.
A menu with user interaction is one of those.
So we can have some button which -when pressed- changes a state in this state manager and when changing a state it can send a number of commands.
It can of course also be done in rules, but that's prone to errors and a lot of work when extending the state machine.
Not only button presses, but also a rotary encoder, etc.

For Nextion displays, you run these on the display self, but I guess it would be very useful for other task automations.
Some state can have its own "member" values, like strings what to display on what position on a display. (e.g. a "page" on the Framed OLED plugin)

tonhuisman, 2021-09-05:
Bitmaps are supported by AdafruitGFX, but they are in an specific format, so I could add something like
`<trigger>,spb,<nr>[,fgcolor[,bgcolor]]`
(spb = show predefined bitmap, I want to reserve `bmp` for actual bmp file support I _might_ add in a separate PR once this is merged). It would be a compile-time option, I guess, as not everyone will need that feature, and it probably won't fit well in the 'size challenged builds'.
We only have to find a usable set of those predefined bitmaps 😉.

HV-NL, 2021-09-05:
> 
> It took a bit of time, but I found the bug, and fixed it. Thanks for pointing it out! As usual, any assumption (I) made is wrong 😄
> (The text size wasn't recalculated when changing the rotation, but the `txs,2` command does that in the example. Using `txs,1` in line 1 would have also 'fixed' it.)
> 

Check! Now my lines are shown as I expected! 
Thanks!

tonhuisman, 2021-09-06:
> how difficult is it to (optionally) make large fonts with high resolution?

I downloaded a font called Amerika sans (from a free fonts site, chose it because it's round and curvy) and converted it in 16pt size into a AdafruitGFX compatible font (.h file). (Had to implement a font-specific vertical offset option, but that was due anyway.)
Then put it on the screen (font scaling is 1, just like the default font showing the IP-address in the 3rd line), and the resolution is quite nice.
This is on a 2" 240x320px display:

[image]

Bigger font sizes are possible, 18 or 20pt would still work out ok, but as we can't scale below a factor 1, it might get too large to be useful. And this font adds 4.2kB to the .bin file.

TD-er, 2021-09-06:
Well 4k for a special purpose build is not too bad.
The fonts for the Framed OLED plugin are also quite large.

So now we need to find a good looking font which does have monospaced numbers (decimal dots can have different size) for showing live updating measurement data and we can make a nice looking instrument.

But this can be done in a later PR :)

tonhuisman, 2021-09-06:
This looks like a nice, fixed width, font: https://www.1001freefonts.com/white-rabbit.font I already have it included in 8pt and 12pt, but adding it in 16pt (maybe 18pt) and 20pt sounds reasonable.

TD-er, 2021-09-06:
Yep looks nice :)
It has a touch of retro and still doesn't look too '80s.
ALso it has a dash through the 0, which is nice.

tonhuisman, 2021-09-12:
The User-define SPI pins feature has been merged, so that is now available in all ESP32 builds.

Here are the updated builds:
[ESPEasy_display_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7150084/ESPEasy_display_ESP32_4M316k.zip)
[ESPEasy_display_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7150085/ESPEasy_display_ESP8266_4M1M.zip)
[ESPEasy_max_ESP32_16M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7150089/ESPEasy_max_ESP32_16M1M.zip) (Includes all fonts, but requires an ESP32 with 16MB flash!)

**Attention**
If you have used a pre-release ESP32 build before, **and** have configured User-defined SPI pins, then first thing to do is select the Hardware tab and re-enable the User-defined SPI setting again (GPIO pins should still be correct), or the ESP will crash when trying to display the Devices list, as the settings-value for this option has changed from previous builds.
**Attention**

Stromer66, 2021-12-07:
is it possible to change fonts in ESP8266 4M1M and ST7735?
i try ~font,freesans~easy but no success.

this plugin is very good! thanks!

tonhuisman, 2021-12-07:
I've created a custom ESP8266 4M1M build and included all available fonts, for you to test, as all fonts won't currently fit in the Display builds (they are only included in the MAX builds, by default, but you'll need an ESP32 with 16MB Flash for that)
[ESP_Easy_mega_20211207_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7671506/ESP_Easy_mega_20211207_custom_ESP8266_4M1M.zip)

And also a pdf of the documentation:
[Display - ST77xx TFT — ESP Easy 2.1-beta1 documentation.zip](https://github.com/letscontrolit/ESPEasy/files/7671516/Display.-.ST77xx.TFT.ESP.Easy.2.1-beta1.documentation.zip)

Stromer66, 2021-12-08:
Hi Ton,

thanks for your fast reply.
in this build the fonts work, but if i use a different font as default in combination by the interval setting, the displayed value will not cleared.
if i coose default font the value displayed correctly.
[image]

tonhuisman, 2021-12-08:
I have seen this behavior when omitting the background color from the command (or using a text sub-command that doesn't have (background) color arguments). Changing the used sub-command could solve that.
When using predefined content in the device configuration, you can change the "Text print Mode" option to "Clear then truncate exceeding message", from the default "Truncate exceeding message".

tonhuisman, 2021-12-08:
I may need to have a look at the way the extra fonts are generated, may have something to do with some kind of 'transparency' in the fonts.

tonhuisman, 2022-03-20:
Keeping documentation for P116 updated in [this comment](https://github.com/letscontrolit/ESPEasy/issues/836#issuecomment-988253502)

PR #3761 (2022-03-31): [P116] ST7735/ST7789/ST7796 TFT Plugin and AdafruitGFX helper
- New plugin P116: ST77xx TFT
  - Supports ST7735s/r 128x128, 128x160, 80x160 pixel TFT 65535 color displays
  - Supports ST7789vw 240x320, 240x240, 240x280, 135x240 pixel TFT 65535 color displays
  - Supports ST7796s 320x480 pixel 65535 color displays
- Introducing `AdafruitGFX_helper` class, as a generic interface for all displays using the Adafruit_GFX library
- Add some optional fonts (1 proportional, 5 fixed width), disabled by default
- Plugin is included in `DISPLAY` builds, that have been set to `LIMIT_BUILD_SIZE` for ESP826 [... cut]
closed 2022-03-31 by TD-er