letscontrolit/ESPEasy #838: Blynk controller

TD-er, 2018-02-09:
A bit the other way around, but this issue is about the "new" Blynk controller, added in PR #837
It is a re-master of PR #92.

So to collect all issues that may arise, use this issue.

uzi18, 2018-02-10:
OK, little investigation:
- it is possible to use your own Blynk server on Rpi, so need to add hostname/port settable
- Blynk launch own local MQTT server, but haven't tested it.

uzi18, 2018-02-13:
After some tests have prepared new pull request #863 

TODO: I think we can fill empty hostname with blynk-cloud.com then it will work out of the box.

sassod, 2018-02-13:
I'm just testing it on a vps  but it seems it is not working, server is
running ok ports are all opened auth token is correct, V0 value is ok even
made a port forwarding from 443 to 9443, tested also with 8443, the server
is listening on all 3 ports

maybe possible issue with my setup of the blynk server??

881452 : GET /b31afdb8f0de4b86988e69a279381221/update/V0?value=25.12
HTTP/1.1 Host: 5.134.115.136 Connection: close
881857 : HTTP : closing connection

COMMAND PID USER   FD   TYPE     DEVICE SIZE/OFF NODE NAME
java    788 root   43u  IPv6 2097680131      0t0  TCP *:9443 (LISTEN)
root@sassod:~# sudo lsof -i tcp:443
root@sassod:~# sudo lsof -i tcp:8443
COMMAND PID USER   FD   TYPE     DEVICE SIZE/OFF NODE NAME
java    788 root   41u  IPv6 2097680129      0t0  TCP *:8443 (LISTEN)

fverschure, 2018-02-13:
Hi, 

@sassod :Please note that the creator of the plugin suggested that this plugin uses the blynk http api. I no not know on which port the http api is running. Did you try to look which application is using the ports which are serving on your rpi?

I just tested it with the blynk cloud service: blynk-cloud.com:80
@uzi18 : good suggestion to replace a blank server value with blynk-cloud.com, would it be also a good idea to set the default port to 80 since this is the default api port for blynk-cloud.com?

Works fine with a DHT22. 
I can see the output in the app. 

I also try to update a switch input, but I cannot see any data transmitted to the blynk service for this device. It is sending it to the same controller (ID1). Does anybody know if this is by design or am I missing something to have the value passed by the controller?

See debug log below:
`DHT  : Temperature: 22.10
DHT  : Humidity: 42.70
EVENT: DHT#Temperature=22.10
EVENT: Processing time:118 milliSeconds
EVENT: DHT#Humidity=42.70
EVENT: Processing time:133 milliSeconds
GET /<TOKENID_REMOVED>/update/V0?value=22.10 HTTP/1.1
 Host: blynk-cloud.com 
 Connection: close

HTTP/1.1 200 OK

HTTP : Success
connection: keep-alive

HTTP : Success
access-control-allow-origin: *

HTTP : Success
content-length: 0

HTTP : Success

HTTP : Success
HTTP : closing connection
GET /<TOKENID_REMOVED>/update/V1?value=42.70 HTTP/1.1
 Host: blynk-cloud.com 
 Connection: close

HTTP/1.1 200 OK

HTTP : Success
connection: keep-alive

HTTP : Success
access-control-allow-origin: *

HTTP : Success
content-length: 0

HTTP : Success

HTTP : Success
HTTP : closing connection
'

uzi18, 2018-02-13:
ports 80 and 8080 works ok.

uzi18, 2018-02-13:
Switch inpout is not supported jet

uzi18, 2018-02-13:
Correct, ports 80 and 8080 works ok.

PR: Switch input support added, port changed to 80.

fverschure, 2018-02-13:
ok, thanks. If needed I can get it running via rules as well. However it would be a great enhancement to have it also send its value to blynk. I understand that updating the value from blynk back to esp-easy would take some additional coding. Without it values would become out of sync, but this can also be implemented via a rule

fverschure, 2018-02-13:
I will also look into the mqtt implementation from blynk to see how this works.

uzi18, 2018-02-13:
@fverschure could You paste here rebuilded firmware with my patch?
You cen fetch variables in rules by use of: BlynkGet cmd.

uzi18, 2018-02-13:
to set: `BlynkGet update/V30?value=7`

uzi18, 2018-02-13:
@fverschure  don't know if mqtt works, it was only information in cfg and log files

fverschure, 2018-02-13:
@uzi18 : Looks like blynk planned to implement mqtt, however on the forum lots of people are asking if it is already implemented and where the documentation is. So not sure on the actual status.

Not sure if if I understand your question with regards to the patch. I understand that you've updated the code. I guess we now need to wait until @psy0rz has integrated it in a new build so we can test it?

uzi18, 2018-02-13:
OK, have testes out with local server - works flawlessly :)

closed 2018-02-27 by TD-er

uzi18, 2018-02-27:
All my test confirm Blynk controller works ok, maybe it will need some features  but for now it is ok.

tyomikh, 2018-10-19:
Hello.
I flashed ESP_Easy_mega-20181017_test_ESP8266_1024.bin
BlynkGet update/V1?value=%syssec% works as expected
BlynkGet 2,1,get/V2 reads the status correctly but causes infinite loop (I'm trying to read button state via rules)

the trace:
		71411: Command: blynkget
		71420: HTTP : C012 connecting to blynk-cloud.com:80
		71473: GET /************************/get/V2 HTTP/1.1^^ Host: blynk-cloud.com ^^ Connection: close^^^^
		71523: HTTP : Success
		71524: HTTP : Success
		71525: HTTP : Success
		71526: HTTP : Success
		71527: HTTP : Success
		71527: HTTP : Success
		72527: Timeout while reading input data!
		72528: HTTP : Success
		72528: Blynk get - get/V2 => 1
		73528: Timeout while reading input data!
		73529: Blynk get - get/V2 => 1
		74529: Timeout while reading input data!
		74530: Blynk get - get/V2 => 1
these two lines are repeated every second. It also seems to hang and doesn't activate new rules (when I change blynk button state to 0 and trigger the same rule the trace is not updated)

My rule:
on getVal do
   BlynkGet 2,1,get/V2 
   event,MYSWITCH=[dummy#switch]
endon

Controller #2 is set as Blynk HTTP [testing] with default settings and correct controller password, hostname=blynk-cloud.com; port=80
I tried to set timeout 100->1000: the same
I tried to change check reply = check ack: it breaks BlynkGet update/V1?value=%syssec% and causes the same infinite loop with the trace above

Device task #2 set as dummy, enabled(makes no difference), simulate data type = <I tried different with the same result>, send to controller = none selected

tyomikh, 2018-10-19:
INFOPAGE:
[image]

DEVICES:
[image]

DETAILS:
[image]
[image]

CONTROLLERS:
[image]

1st one:
[image]

2nd one:
[image]

RULES 1-2 are just tested to switch on/off relay by schedule:
RULE # 3 is empty, RULE # 4:
on Rules#Timer=2 do
   TaskValueSet 2,3,%syssec%
   TimerSet,2,30
endon

on remoteTimerControl do
   GPIO,5,%eventvalue%
   BlynkGet update/V1?value=%syssec%
   BlynkGet update/V3?label=%sysyear%-%sysmonth%-%sysday%__%systime%
endon

on getVal do
   BlynkGet 2,1,get/V2 
   event,MYSWITCH=[dummy#switch]
endon