letscontrolit/ESPEasy #842: What are the chances of support for SSD1309 OLED controller?

Budman1758, 2018-02-10:
I just acquired one of these displays.  

[https://www.ebay.com/itm/RGB-White-2-42-OLED-LCD-Display-SSD1309-128x64-SPI-Serial-Port-for-Arduino-C51/172967178530?ssPageName=STRK%3AMEBIDX%3AIT&var=471729716998&_trksid=p2060353.m2749.l2649](url)

I could have swore that I was getting a display with a SSD1306 chip but I guess I'm getting dyslexia or something cause I now have a 20 dollar paperweight.  I realize you Dev type folks are a bit busy but I just thought I'd ask.........

sassod, 2018-02-10:
> SSD1309-128x64-SPI-Serial-Port-for-Arduino-C51/
> 172967178530?ssPageName=STRK%3AMEBIDX%3AIT&var=
> 471729716998&_trksid=p2060353.m2749.l2649 <http://url>

I dont understand why you would spend 20$ on a graphic display when you can
get a lot more for that amount of money. You could get a nextion lcd 3.2"
with touchscreen. The plugin is in the playground and it works at lesat for
me with a ESPEasy_v2.0.0-dev11, haven't tried compiling latest builds but I
think it should work.

Budman1758, 2018-02-10:
Well..... Had I known...  Like I said, it was an error that I got that particular display.

TD-er, 2018-02-10:
Those nextion displays are quite different.
You have to create some template in their software and then act on events from that display.

TD-er, 2018-02-10:
Oh and I don't have one (yet ;) )

Grovkillen, 2018-02-10:
Maybe get that playground plugin for Nextion into the official release?

TD-er, 2018-02-10:
Hmm I just needed an excuse to buy one ;)

s0170071, 2018-02-10:
We should make it a habit to get it for free from the feature requester...
Along with some chocolate and two beer 😁
-- 
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

Grovkillen, 2018-02-10:
@TD-er I got two... you want one? 👍

sassod, 2018-02-10:
I'm more than willing to provide a display for the developers if that means
that the display is going to be considered supported in the future
releases. 20 or 30$ for something that may benefit the requester is a small
price for it.

Tell me where to send and i will order one for you... 😁

TD-er, 2018-02-10:
I already ordered one this morning.
But it's coming from Banggood, so it may take 7-20 days to arrive.

Budman1758, 2018-02-10:
@s0170071   Let me know where to send it and it will happen.  :>)
I would rather keep using my previous paperweight.........

pastaclub, 2018-03-06:
Not sure whether you're aware of it, but there is a discussion [here](http://www.letscontrolit.com/forum/viewtopic.php?t=2971) where someone made an SSD1309 display work with the SSD1306 code by exchanging 3 resistors.

Other than that, the Nextions were mentioned. I worked with them in the past and it's quite different. You have to use their editor to build a GUI, then flash that to the Nextion and communicate with it through a serial port. While the bang for the buck is quite good, their editor software has a crappy usability and its parser forces you to write ugly code. Changing your application means not only changing your microcontroller code but the Nextion GUI as well, which means re-compiling that and flashing it to the device. And then it's a TFT (which gets hot and has inferior contrast compared to an OLED), and the touchscreen is nothing like a smartphone - there is a foil on the display that you have to depress, like TomTom navigation devices 15 years ago. If you need nice colored bitmaps, go for the Nextion, I'm just saying there are also good reasons to go with a monochrome OLED.