letscontrolit/ESPEasy #843: MQTT message retain state doesn't get saved

lativl, 2018-02-10:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
1. Enable option "MQTT Retain Msg" in Advanced menu
2. Submit
3. Execute command in rules (like " Publish home/alarm,disarmed")
4. Check message in MQTT client (retained=false)

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes

### Expected behavior
Tell us what should happen?
no

### Actual behavior
Tell us what happens instead?
MQTT message sent with flag "retained"

### System configuration

Hardware:

Software or git version:

TD-er, 2018-02-10:
The retained flag is actually stored correctly within ESPeasy.
Up-to the point where the PubSubClient::publish() is being called, the retained flag is still how it should be.
The bug seems to be in the PubSubClient library.

I will continue digging.

TD-er, 2018-02-10:
Ah, found it... 
It is actually working just fine :)

The retain bit is just a bit for sending to the broker.
The broker will then keep the last message in that topic with a retain flag. As soon as someone subscribes to that topic, it receives immediately the last retained message.
That's useful when the sender has some large interval for sending values.

So send a MQTT message with 'retain' flag set and a long enough interval to the broker.
Then -before a new message is sent- subscribe to that topic and the first message you'll get is marked 'retained'. All later messages are not marked 'retained', since you get them immediately from the broker.

For more information: https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages

As 'proof' the two messages I received: (bottom one received first)
[image]

Using these rules:
```
on System#Boot do
  timerSet,1,10
endon

On Rules#Timer=1 do
  Publish %sysname%/IP,%ip%
  timerSet,1,30
endon
```

lativl, 2018-02-11:
Yes, it seems you are right. Thanks for your efforts and good analysis!

closed 2018-02-13 by lativl