letscontrolit/ESPEasy #846: Improvement to _P042_Candle.ino

Nornode, 2018-02-10:
### Expected behaviour
Possibility to adjust LED stip/preferences, not just lighting schemes.
On starting/adding the output preferences should be to edit the following for the plugin:
#define **NUM_PIXEL       20**         // Defines the amount of LED Pixel
#define NUM_PIXEL_ROW    5         // Defines the amount of LED Pixel per Row
#define RANDOM_PIXEL    70         // Defines the Flicker Level for Simple Candle
#define BRIGHT_START   128         // Defines the Start Brightness
#define BASE_TEMP       21         // Defines the Base Temp for TempRange Transformation

I would also expect it be possible to adjust the setting by the normal 
"http://XX.XX.XX.XX/control?cmd="
But that might be a longer endeavour?
--> nothing about it here: https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P042_Candle.ino

### Actual behaviour
With LED strips longer than 20*3 LEDs you don't control all of your leds.
It resulted in me thinking my 5m (100*3) strip was broken...

### System configuration
Hardware: NodeMCU

Software or git version:
ESPeasy Mega 20180209

Grovkillen, 2018-02-12:
Maybe even make that more generic since given (total) amount of pixels you may accidentally put pixels per row not compatible with the total number of pixels. Like this:

```
#define NUM_PIXEL_COL 4
#define NUM_PIXEL_ROW 5
```

flexiti, 2018-03-28:
Is there a simple way to control the Candle using the MQTT protocol?

Nornode, 2018-03-28:
There's something commented out on the lines 368-383 about both MQTT and API requests.
I did not get around to test it yet.

flexiti, 2018-03-28:
They do not work with MQTT

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er