letscontrolit/ESPEasy #853: favicon.ico

s0170071, 2018-02-12:
sometimes, Firefox just keeps asking for favicon.
Doesn’t it get a 404 ?

In this situation I also get 0.0.0.0 permission denied errors and the web interface 'hangs'
Seems connected somehow.

TD-er, 2018-02-12:
All unhandled requests should return a 404, or a permission denied.
I don't know why it then looks like a connection from 0.0.0.0.
Or maybe it is a req. from localhost, which is not correctly mapped to 127.0.0.1?

Grovkillen, 2018-02-12:
I guess you could add your on favicon, but yes it shouldn't complain.

s0170071, 2018-02-14:
There is apparently no answer to the favicon request. 
If you enter in your browser 
http://192.168.1.36/blabla - no answer
http://192.168.1.36/blabla.esp - not found error (.esp files are handled)
http://192.168.1.36/favicon.ico - no answer

TD-er, 2018-02-14:
Do we have a nice ESPeasy logo?

Grovkillen, 2018-02-14:
Yes we do ;)
https://imgur.com/a/wPJOB

TD-er, 2018-02-14:
Done :)

s0170071, 2018-02-15:
Nice. Thanks.

closed 2018-02-27 by TD-er