letscontrolit/ESPEasy #856: illegal instruction

s0170071, 2018-02-12:
while playing with the free memory (I was actually reserving 17k just for fun) I came to the point where configured  a new framed oled display. With only 1k free ram left, the inevitable crash followed when I pushed submit. Since then the ESP keeps crashing on startup. Might be the config file being corrupted. 

Exception decoder shows: 

```
Exception 0: Illegal instruction
Decoding 28 results
0x4021c113: Plugin_036(unsigned char, EventStruct*, String&) at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x4010020c: _umm_free at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x40206b31: spiffs_phys_rd at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_cache.c line 153
0x40204204: spiffs_page_data_check at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_nucleus.c line 1164
0x402065bc: spiffs_object_read at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_nucleus.c line 2037
0x4010053d: _umm_realloc at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1495
:  (inlined by) realloc at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1715
0x40243407: String::changeBuffer(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x40243453: String::reserve(unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x40243485: String::copy(char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x402434d2: String::String(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x40243485: String::copy(char const*, unsigned int) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x4020fc6d: PluginCall(unsigned char, EventStruct*, String&) at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x40210e00: StringSumHelper at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
:  (inlined by) FileError(int, char const*) at /home/john/Arduino/scetchbooks/ESPEasy/Misc.ino line 538
0x402434d2: String::String(char const*) at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 714
0x40100690: free at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1739
0x4022b8fe: SensorSendTask(unsigned char) at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x40100201: _umm_free at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1281
0x40271a8a: wifi_get_opmode at ?? line ?
0x4010020c: _umm_free at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x40100690: free at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1739
0x40209bcf: timePassedSince(unsigned long) at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x4022ba24: checkSensors() at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x402305cb: runOncePerSecond() at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x40209bcf: timePassedSince(unsigned long) at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x40230760: loop at /home/john/Arduino/scetchbooks/ESPEasy/TimeESPeasy.ino line 147
0x402443c4: loop_wrapper at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /home/john/ArduinoPortable/arduino-1.8.5_ESPgit/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109
```

s0170071, 2018-02-12:
putting a return statement at the top of the framed oled display allowed the ESP to boot up. 
 
```

  int linesPerFrame;            // the number of lines in each frame
  int NFrames;                // the number of frames
return true;
  switch (function)
  {

```

This allowed me to pull the config file to play with... Many zeros in it. But still, it shouldn't crash the ESP.

[config_ESP_Easy_U0_Build20100_201808.dat.zip](https://github.com/letscontrolit/ESPEasy/files/1715652/config_ESP_Easy_U0_Build20100_201808.dat.zip)

TD-er, 2018-02-12:
There is also something I still want to look at.
Not sure if it is reproducible.

- At position N, add OLED Framed plugin.
- At position N+1, add some plugin (I used system info showing the free ram) and git that plugin some name.
- Delete the OLED framed plugin.

Is the name of the next plugin still there?

s0170071, 2018-02-12:
Yes, the name stays.

TD-er, 2018-02-12:
OK, good to know.
But still I guess we should (add a?) check whether datablocks don't overwrite eachother in the settings.

s0170071, 2018-02-12:
I would rather have the check when applying the corrupted data to the plugins. 
If memory is low, the ouput of the config file is unpredictable. And so is program execution in general. Any checks too.

In my case, even the checksum was ok.

TD-er, 2018-02-12:
True, detection of implausible data (since CRC is correct, it is 'correct' data) should be done more often.
And also we might want to save to new files and mark some as "fallback" or "default" to be able to recover from some corruption, or inadvertently set unusable values.

TD-er, 2019-10-27:
This issue is no longer relevant, since all mentioned points are taken care of now.

closed 2019-10-27 by TD-er