letscontrolit/ESPEasy #857: Framed Oled Submit is memory intensive

s0170071, 2018-02-12:
Tracking the memory I saw that when saving the configuration of an already existing  framed oled display 3kB of memory are used. 
Here are the logs from 

- boot
- click devices
- click edit. 
- click submit. 

The lowest memory is registered by the webserver, however, it is not caused by it. In fact, the webserver will always see the lowest memory since it always comes on top.
Should we separate webserver memory monitoring from all other monitoring ?

```

INIT : Booting version: (custom)
INIT : Warm boot #12
FS   : Mounting...
FS   : Mount successful, used 151604 bytes of 2949250
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:21536
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Connecting MNET attempt #1
WIFI : Connected! IP: 192.168.1.36 (ESP_Easy_11)
WD   : Uptime 0 ConnectFailures 0 FreeMem 15984
```
click on "Devices"
```
sendcontent free: 15184 chunk size:173
sendcontent free: 15104 chunk size:584
sendcontent free: 15296 chunk size:378
sendcontent free: 15680 chunk size:397
sendcontent free: 15280 chunk size:418
sendcontent free: 15680 chunk size:394
sendcontent free: 16008 chunk size:397
sendcontent free: 15976 chunk size:390
sendcontent free: 16016 chunk size:283
sendcontent free: 16016 chunk size:0
Ram usage: Webserver only: 1448 including Core: 3520
```
now edit:

```
sendcontent free: 15472 chunk size:173
sendcontent free: 15472 chunk size:584
sendcontent free: 15664 chunk size:378
sendcontent free: 16048 chunk size:397
sendcontent free: 15648 chunk size:418
sendcontent free: 16048 chunk size:394
sendcontent free: 16352 chunk size:452
sendcontent free: 16192 chunk size:446
sendcontent free: 15976 chunk size:411
sendcontent free: 15888 chunk size:417
sendcontent free: 15888 chunk size:421
sendcontent free: 15568 chunk size:406
sendcontent free: 15568 chunk size:415
sendcontent free: 16376 chunk size:470
sendcontent free: 16384 chunk size:395
sendcontent free: 16384 chunk size:0
Ram usage: Webserver only: 1048 including Core: 3120
```
now submit.

```
sendcontent free: 13280 chunk size:173
sendcontent free: 13280 chunk size:584
sendcontent free: 13472 chunk size:378
sendcontent free: 13856 chunk size:397
sendcontent free: 13456 chunk size:418
sendcontent free: 13856 chunk size:394
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved config.dat
FILE : Saved security.dat
sendcontent free: 14160 chunk size:452
sendcontent free: 14000 chunk size:446
sendcontent free: 13784 chunk size:411
sendcontent free: 13696 chunk size:417
sendcontent free: 13696 chunk size:421
sendcontent free: 13376 chunk size:406
sendcontent free: 13376 chunk size:415
sendcontent free: 14184 chunk size:470
sendcontent free: 14192 chunk size:395
sendcontent free: 14192 chunk size:0
Ram usage: Webserver only: 1048 including Core: 3120
```

TD-er, 2018-02-12:
I've tried to map what is stored where: Wiki: [Settings Storage Layout](https://www.letscontrolit.com/wiki/index.php/Settings_storage_layout)
The OLEDframed plugin uses `CustomTaskSettings`
Those custom task settings are always read from the flash and there is only one instance of it in memory.
So every time it is being used, it is read from flash.

It can be that loading (and saving) this will read all using sizeof() that struct. 
My guess is that this may cause a lot of issues and perhaps using a single String, or String[12] may help a lot.

Grovkillen, 2018-03-23:
Can this be closed?

TD-er, 2018-03-23:
I guess so, since the current implementation of the plugin no longer tries to read from flash each time.
However, when the plugin data is being saved or accessed, the settings are read from the flash. This may still be rather memory hungry, compared to other plugins.

Grovkillen, 2018-04-06:
I close this one now. Good job @TD-er

closed 2018-04-06 by Grovkillen