letscontrolit/ESPEasy #858: security concern : stealing wifi password

Kurairaito, 2018-02-12:
### Steps to reproduce
How can we trigger this problem?
Configure you firmware like always setting a wifi ssid and a wifi password. for extra protection put a config password.
Then connect throught serial connection and type settings. Wifi ssid and password are revealed. the config password is not asked.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
It happen everytime

### Expected behavior
Tell us what should happen?
No wifi password should be revealed + accessing the setting should be only possible after entering the password.
As this password could be theorically bruteforced, wifi password should never be accessed from serial connection. It should be only possible to set a new password, erasing the older.

### Actual behavior
Tell us what happens instead?
Wifi password is accessible to anybody having a physical access to the esp. this should not be possible as thoses modules can be put outside to grab external informations (wheater for example).

### System configuration

Hardware:
Wemos D1 mini

Software or git version:
v2.0-20180123

TD-er, 2018-02-12:
You could also read the stored files from SPIFFS and there is the password
stored in plain text.
When you have access to the device, you can access the SPIFFS.

Maybe it is a good idea to somehow scramble all data in the security file.

---
Verzonden vanaf laptop

M0ebiu5, 2018-02-12:
If someone has physical access to the esp, he can read all data from flash. Also for this person it will be no effort to descramble the password.

The esp8266 has just not enough resources to be a full scrambled, ssl secured, high security device.

The easiest way to gain more security is to put the esp's in their own firewalled wifi network. Additionally you can implement something like a heartbeat, to detect abnormalities (stolen, modified, etc).

Kurairaito, 2018-02-12:
I understand that anyone having access to the esp can do anything with it, but I think there is a gap between : 
- pluging an usb cable on a computer, typing a single command and get the password in plain text
and
- doing electronics to read all the flash, or doing some real hacking to get it

no?

TD-er, 2018-02-12:
Sure there is some difference, but it has nothing to do with electronics.
Just connect it, open Arduino IDE and read SPIFFS.
Then you'll have a binary with the very obfuscated name like "security" or something like that.
View it with a hex editor and you'll have the WiFi password.

But I agree, we could do a little bit more to hide the WiFi passwd.

Kurairaito, 2018-02-12:
yeah, I removed the electronic part, as I remembered that we don't have to wire anything more to read the flash.

Grovkillen, 2018-02-13:
We might add a section on the wiki with all the precautions that can be made to make stealing your credentials as hard as possible. But like we said, this is not a industrial grade unit. 

ESP8266 probably was designed with the aim to enable IoT for "stupid" products around the house. We're actually doing a lot more than what the engineers at Espressif ever imagined to be done using their hardware. :)

TD-er, 2018-02-13:
We should look into security for the next release (past v2.0)
Also SSL/TLS-1.1 can be used with the next core lib

M0ebiu5, 2018-02-13:
Great! 
Will there be also more than 12 tasks and for each its own mqtt publish topic, multiple mqtt subscriptions and a lot of free ram? :)

TD-er, 2018-02-13:
Yes, for each ESP board a set of those should not be a problem ;)

closed 2019-10-11 by Grovkillen