letscontrolit/ESPEasy #861: safe mode boot

s0170071, 2018-02-13:
inspired by #856 I thought we may implement a safe boot.
- kicks in after the 5th warm boot, 
- ignores all settings.

TD-er, 2018-02-13:
Perhaps also with an interval limit?
For example 5 times within 5 minutes?

s0170071, 2018-02-13:
if that is possible. I don't know how much or even which ram survives the boot.

s0170071, 2018-02-13:
Another thought. We could apply a backup (previous version) of the settings. Whatever settings crashed the ESP were undone...

TD-er, 2018-02-13:
Backup settings or settings marked as "known to work" is always a good idea.

s0170071, 2018-02-13:
We do that after the 5th boot, and discard all settings after the 15th. In case of spiffs is corrupt. 
Speaking of spiffs, is it possible to select distinct sectors for writing ? If the puya chip is worn, we pick another sector. Our settings file is so tiny compared to the spiffs area...

TD-er, 2018-02-13:
The wear-out of PUYA chips is worth another issue: https://github.com/esp8266/Arduino/issues/4061#issuecomment-365139415

SPIFFS does do wear-leveling, but not on PUYA chips.
Wear leveling is by design of SPIFFS.
So I agree that we should add an option to mark one settings file as "known to work" and start writing to a new file with a suffix in the filename for "experimental" changes.  Then you don't have to rename the existing file when marking it as one to keep.
Not sure if that's the right strategy, just a thought.

s0170071, 2018-02-13:
This needs to be a no-brainer with no interaction. You cant interact when things go south...
This may work: 
1. The settings file is written pretty often. We make a backup only if there is not another backup persent.
2. make a backup of the settings when:  
- there are changes. 
- there is not already another backup.
       
3. on next boot monitor the uptime. If it exceeds 1 minute, the settings are not corrupt. Delete backup.

TD-er, 2018-02-13:
Maybe always add a number (single digit?) to the filename and only delete older ones when the SPIFFS gets full, or maximum number of versions is exceeded?
Then start loading the one with the last version. If fail, one step back.

When using a single digit, there should be a gap in numbering to detect the oldest one. (right after the gap)

s0170071, 2018-02-13:
That would require that you rename them all each time you save the config...
But yes, we could do that. 
When the ESP finally comes up delete all config versions that failed.

TD-er, 2018-02-13:
Nope, just append a number on write. Never rename.

e.g.
existing versions with a limit of 5 versions to store:
1, 2, 3, 9, 0
Then 3 is the newest version, 9 the oldest.

With a single digit as number, you can only store up-to (10 - 1 = ) 9 versions, because you need the 'gap' to see which is the oldest.
You cannot use timestamps, since not all will have proper NTP.

s0170071, 2018-02-13:
Ahhh. Now I get it. I was interpreting the gap as a space character in the filename...
Tricky you... ;-)

TD-er, 2018-02-13:
If I can avoid it, I will not use spaces in filenames ;)

mattlward, 2018-02-13:
Will this affect the way a unit behaves if it is away from wifi for to long?  I use one in my truck and it is only connected about half  to 2/3rds of a 24 hour cycle.

TD-er, 2018-02-13:
Not being connected to WiFi is no reason for a reset by watchdog.

Grovkillen, 2018-10-23:
@TD-er would this be considered closing time since your new boot-recovery loop is doing a really great job? Open if you dis-agree.

closed 2018-10-23 by Grovkillen

TD-er, 2018-10-23:
At this moment only single plugins/controllers are disabled, but there should still be a check for "short uptimes" and the rules are not disabled when failing.
So the recovery could be extended. But there is another issue for that.