letscontrolit/ESPEasy #862: Add .bin file name of last flash to the info page/JSON page

Grovkillen, 2018-02-13:
In order to help users to pick correct flash file when doing a OTA it would be nice to have the exact file name which the current FW came from.

I could also make use of it when multi-update many units OTA if the file name was added to the JSON page.

Future releases could even parse the name and compare it with the user picked one and warn if the flash size/version/etc didn't follow the same nomenclature.

psy0rz, 2018-02-13:
Embedding the filename could be done in the same way the MD5 is embedded by crc2.py.

TD-er, 2018-02-13:
Hmm, that will make it very EASY for everyone ;)

Grovkillen, 2018-02-14:
Well, yes. That would be the same. Only difference is that I could upload a custom bin and that would not be included in that info (only "custom" for Git Build). But maybe I could live with that.

Grovkillen, 2018-02-15:
#877 is a good example where only size and FW flavor is not enough.

Grovkillen, 2018-03-08:
The current implementation is listing the generic file name. 
[image]

Grovkillen, 2018-03-08:
We need to have the renamed file listed.

Grovkillen, 2018-03-12:
Is this something that can be done?

s0170071, 2018-03-12:
someone with access to the build process could do it...

Grovkillen, 2018-03-15:
@psy0rz can you make this happen?

Grovkillen, 2018-04-03:
Latest version doesn't even have the "firmware.bin" added to the INFO page... md5 is also failing.
[image]

Oxyandy, 2018-04-03:
Build is a variable, but between firmware versions it doesn't change, 
I think it would be better to put BUILD_GIT into Build and would be really nice to be able to have %build% available.
In other words there are 2, just make it one.

Oxyandy, 2018-04-03:
Actually there are 3 "Binary File Name", "Build Git" & "Build"
If "Binary File Name" (Or just one, which is up to whoever) is hard-coded at compile time, then only use that.
a "Binary File Name" (for example) which is descriptive enough is plenty, why have 3 ?

Grovkillen, 2018-04-04:
I thought I made that clear with my reference to #877 but maybe not. The bin is actually King here since it's the only thing that is pointing to what's actually in the unit code wise. The rest is only describing source code and not the compiled code. Since we already have information regarding the source code I don't see any reason to remove that but compile code is much needed.

Oxyandy, 2018-04-04:
json
```
{"System":{
"Build": 20000,
"Unit": 2,
"Uptime": 1055,
"Free RAM": 23072
},
```
"Build": 20000 <-instead of as is currently - Something more identifiable ?

Grovkillen, 2018-05-13:
This is now implemented

closed 2018-05-13 by Grovkillen