letscontrolit/ESPEasy #864: Devices split over three pages

s0170071, 2018-02-13:
can we display all devices on one page ? It is currently split into three pages showing four devices each. 
What was the reason to do it this way ? Size of the webpage ?

psy0rz, 2018-02-13:
yeah that WAS the reason but we can probably change it once we get your streaming stuff in:)

psy0rz, 2018-02-13:
this is actually a duplicate of another issue, but cant find it now :)

s0170071, 2018-02-13:
I was hoping you 'ld say that. If I find the time tomorrow I'll add it to the streaming PR ? 
Or better keep it seperate ?

psy0rz, 2018-02-13:
just add it to the PR, i can always cherry pick the individual commits. try to keep different "steps" in the development process in different commits, when needed. :)

s0170071, 2018-02-13:
done. 
That was easy :-)

psy0rz, 2018-02-13:
lol thanks :)

closed 2018-02-13 by psy0rz