letscontrolit/ESPEasy #866: Disable serial logging when serial not connected

TD-er, 2018-02-13:
As suggested here: https://github.com/letscontrolit/ESPEasy/issues/847#issuecomment-365406879
Apparently disable logging to serial when not connected will make the ESP response more smooth, use less memory and will lead to world peace.
If we could achieve at least 2 of these, it is worth to try and auto detect whether the serial port is being read and stop sending logs to serial when it is not being read.

uzi18, 2018-02-13:
> As suggested here: #847 (comment)

Try to disable logging at the end of setup()
We need some information at the beginning about ip etc.
Add checkbox in advenced settings with logging on/off,  but for future we
need more extended logs,  to be able to focus on some parts at debugging
time.

TD-er, 2018-02-13:
The logging is something that really needs some attention.
I was thinking about adding some bit-sets to enable/disable certain categories, like protocol, info, error, debug, etc.
Now it is just a filter and when set high enough, you will get all.

But that's another issue.

For now I am trying to detect whether the FIFO is full.
There is this function `Serial.availableForWrite();`
But it is always returning 128, regardless the state of the serial port.

So have to look a bit further.
I guess I will commit what I have for now, so others can also take a look (added the output of this test to the sysinfo page)

s0170071, 2018-02-14:
- Less

 memory - that could be done. Just like with the web server I would try and stream out logs directly instead of building up strings. Problem is that this is done throughout all plugins. So this is quite something.

- Detecting

 a connection may not be possible as from the ESP point of view there is always the USB transceiver connected to it. Whatever pin push and read back magic one would try- unless you get sent back anything from the host I don't think you can auto-detect this.

- I would go for
 
disabling serial logging at a certain time after boot. Devs look at it for like two minutes and after that... re-flash. And world peace- were you thinking about a plugin or a more instant solution?

TD-er, 2018-02-14:
Wait with streaming logs.
I was also planning to detect whether it is even needed to generate a log string.

And also allow for proper filtering.

Detection is still an unsolved problem

The last item is also a bit tricky.
Maybe you could fit your streaming solution to do it?
Otherwise I think we have too little resources on the ESP to manage.

TD-er, 2019-10-27:
Current implementation does have some buffer for the log to serial, so logging is no longer blocking
It is impossible to detect whether the serial port is actually read.

So this issue can be closed, since the original issue cannot be implemented and we now have a better fix for the actual problem.

closed 2019-10-27 by TD-er