letscontrolit/ESPEasy #869: Domoticz Controller sends non JSON Format

jannnfe, 2018-02-14:
### Steps to reproduce
If I activate the controller Domoticz MQTT and want to send my temperature and humidity from the DHT22 sensor or PPM from the MQ-9 sensor to Domoticz, a "Connected" or "Connection Lost" in "domoticz/out" will be sent in non JSON format. This causes problems with the use of the Homebridge plugin eDomoticz because it does not work as long as the MQTT topic does not contain all messages in the JSON format. This message is sent every time the controller connects or breaks the connection. I do not know if that is the case with other plugins, but I noticed with these two plugins:
Environment - DHT11/12/22 and Analog input - internal

Please have a look at [this](https://github.com/PatchworkBoy/homebridge-edomoticz/issues/129#issuecomment-365458148) Issue from eDomoticz.

### Expected behavior
There should be no status message when connecting or disconnecting in non JSON format in the "domoticz/out" topic.

### Actual behavior
The controller sends a "Connected" or "Connection Lost" to the "domoticz/out" topic in non JSON format, which causes an error with the use of the Homebridge plugin eDomoticz.

### System configuration
Hardware:
ESP8266 Wemos D1 Mini
DHT22
MQ-9 Gas Sensor

Software or git version:
mega-20180114
P005 & P025
Domoticz MQTT Controller

TD-er, 2018-02-14:
Does this also apply to other MQTT controllers, like OpenHAB?

So should we add a check mark to force json on all messages?

jannnfe, 2018-02-14:
I looked into the Messages with different Controllers with:
```
mosquitto_sub -v -t '#'
```
Tested with OpenHAB Controller (First Line) and Domoticz Controller (Second Line):
```
ThermostatSensor/status Connected
domoticz/out Connected
```

For example Sonoff-Tasmota sends a different MQTT message for this "Debug" message (Online, Offline instead of Connected, Connection Lost):
```
tele/Thermostat/LWT Online
```
Maybe this could be a possible solution but a checkbox to force JSON output messages could solve the problem as well (i hope :D).

TD-er, 2018-02-15:
Just to make sure I get it... Does Tasmota send to a different topic? Or is it just that they send another word?

TD-er, 2018-02-15:
Is it possible the Offline message of Tasmota is not a regular message, but a "Last Will and Testament" message sent by the broker?

jannnfe, 2018-02-15:
I'm sorry, I'm not sure, but I think it's a "Last Will Testament", because that's what it's called "tele/Thermostat/LWT". I do not know exactly yet with MQTT and I'm still learning about things, etc., but noticed the problem with the non JSON format and wanted to know if there might be a solution for it. Sorry, please keep me up to date if there is any news on this topic.

TD-er, 2018-02-15:
I was just reading about MQTT (all a bit new to me too) and saw this on the [TASMOTA wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features).
Tasmota uses several topic prefixes like

- cmnd
- tele
- stat

And given that the "LWT" part is still part of the topic (separated with a '/' thus still part of the topic), to what topic(s) are you subscribed?
Maybe we should send the controller status to a "LWT" (sub)topic?

Currently we don't have support for LWT, but that's probably not too hard to implement.

jannnfe, 2018-02-15:
Yes the Controller Status by example Topic: devicename/controller/LWT
And the sensors as they are now, so that only messages in JSON format are sent.
Sorry for my beginners expressionist but you probably know better how to do it best! 👍

TD-er, 2018-02-15:
Well I don't know a lot about MQTT, also just learning. It appeared asking google for "MQTT essentials" is quite a good keyword ;)

jannnfe, 2018-02-15:
But you will still look for the problem? I would be glad!
In the meantime, what can I do myself to temporarily solve the problem? Is it possible to switch off the statuses somewhere? Thank you!

TD-er, 2018-02-16:
I still working on it ;)
And I found some other issue as well.

There will be a pull request within the hour I guess and not sure when @psy0rz will start merging, but I guess he will do that tomorrow evening so it will be in the next build (Saturday morning) I guess.
But if you can build it yourself, you could try to build the PR I will make.

jannnfe, 2018-02-16:
Okay Nice thank you for your quick help. Thats super cool! :)

TD-er, 2018-02-16:
You can test it if you like.
I created a PR based on branch v2.0
@psy0rz will merge it later on to the Mega branch.

jannnfe, 2018-02-16:
Thank you for your help. I will try it tomorrow :)

closed 2018-02-19 by jannnfe

TD-er, 2018-02-19:
Closing the issue, so I can conclude the fix works as expected?

jannnfe, 2018-02-19:
Oh yes sorry, my fault. It works perfectly!
Thanks for your nice work :)

1043717432, 2018-04-25:
@jannnfe   hi,How does mq9 intervene in espeasy, what do you choose in Devices?