letscontrolit/ESPEasy #870: rules delay to post vcc on deepsleep

rene9900, 2018-02-14:
i made a topic at letscontrolit forum, and was asked to post the url here since it may be some fault in the firmware, https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=4824

i have a eps8266-01 running ESPEASY 120, red led cut off for lowest power consume possible, it wake up on reset, and i ordered some LIPO's on ebay, want to try see how long it runs on battery. problem i have now is to make it publish VCC waking up then going to sleep, it works perfect if deep sleep is disabled but not if it is on, then all it publish is the online message, i made rules for delaying the boot so it has time to read the vcc correct..this is my rule set, can any of you locate the "problem"?
[image]

uzi18, 2018-02-14:
Did You try to use  Delay command instead Timer, and just after Delay Publish in System@Boot?

rene9900, 2018-02-14:
i just tried delay command, changed rules to this:

On System#Boot do
Publish /%sysname%/status,%sysname%_Online
gpio,15,1

delay 8000
Publish /%sysname%/VCC,%vcc%
Publish /%sysname%/Time,%systime%
endon

result is: 

EVENT: System#Boot
ACT  : Publish /8266-5/status,8266-5_Online
ACT  : gpio,15,1
SW   : GPIO 15 Set to 1
ACT  : delay 8000
MQTT : Topic: /8266-5/status
MQTT : Payload: 8266-5_Online
ACT  : Publish /8266-5/VCC,-1.00
ACT  : Publish /8266-5/Time, 0:00
NTP  : NTP sync requested
NTP  : NTP send to 195.137.195.250
NTP  : NTP replied!
MQTT : Topic: /8266-5/VCC
MQTT : Payload: -1.00
MQTT : Topic: /8266-5/Time
MQTT : Payload: ,0:00
SYS  : 4.47
EVENT: VCC#VCC=4.47
Enter deep sleep...
state: 5 -> 0 (0)
rm 0
del if0
usl

Grovkillen, 2018-02-14:
Try this?

```
On System#Boot do
Publish /%sysname%/status,%sysname%_Online
gpio,15,1
Publish /%sysname%/VCC,[VCC#VCC]
Publish /%sysname%/Time,%systime%
endon
```

uzi18, 2018-02-14:
Try also delay with 25000 as vcc is not measured.

uzi18, 2018-02-14:
> i tried
> Got Result:
> ACT : Publish /8266-5/VCC,0.00
> NTP : NTP send to 95.154.26.34
> still no vcc or system time :(

Use delay 25000

rene9900, 2018-02-14:
@25000 it's basicly the same :( i also tried with a esp8266-12 chip on a breadboard to test if it was the -01 model based error, but it have same issue

EVENT: System#Boot
ACT  : Publish /8266-5/status,8266-5_Online
ACT  : gpio,15,1
SW   : GPIO 15 Set to 1
ACT  : delay 25000
MQTT : Topic: /8266-5/status
MQTT : Payload: 8266-5_Online
pm open,type:2 0
ACT  : Publish /8266-5/VCC,-1.00
ACT  : Publish /8266-5/Time, 0:00
NTP  : NTP sync requested
NTP  : NTP send to 192.36.143.130
NTP  : NTP replied!
MQTT : Topic: /8266-5/VCC
MQTT : Payload: -1.00
MQTT : Topic: /8266-5/Time
MQTT : Payload: ,0:00
SYS  : 4.47
EVENT: VCC#VCC=4.47
Enter deep sleep...
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl

uzi18, 2018-02-14:
@Grovkillen fix provided, but as usual someone need to test it :)

uzi18, 2018-02-14:
With this patch it is not needed to use Rules anymore,
just for special cases we provide System#Sleep event.

rene9900, 2018-02-15:
correct me if im wrong, but i just put in the code on my arduino ide inside the ESPeasy.ino in the void loop right?
im getting compile errors on 
ESPEasy:683:27: error: 'isDeepSleepEnabled' was not declared in this scope
    if (isDeepSleepEnabled())
                           ^

uzi18, 2018-02-15:
Travis said everything is fine :)

uzi18, 2018-02-15:
@Grovkillen  @TD-er  is it fine for You?

TD-er, 2018-02-15:
I will look at it this evening.

uzi18, 2018-02-15:
Have some new ideas for rules processing improvement but need to find solution to compile ESPEasy in eclipse or Arduino IDE.

TD-er, 2018-02-15:
Compiling in Arduino IDE is not that hard.
Just make a symbolic link (is also possible in Windows) to the src dir with "ESPEasy" as name and open that in the Arduino IDE.
Then also copy the libraries folder to the lib dir of your portable Arduino sketch directory. 
Use portable installation, or else you'll get conflicts with other libraries installed in the Arduino IDE.

uzi18, 2018-02-15:
Need to install esp8266  v 2.3.0 or something else?
How You define to build with plugin testing/development?

I dont want to mess with my libs used in rest of my projects.
Arduino builder has got some improvements waiting in PRs to generate cmake
independent project from arduino sources, will try it.

rene9900, 2018-02-15:
will the ESPeasy.ino file be up for download? apparently i cant figure out how to change it :(

uzi18, 2018-02-15:
go here: https://raw.githubusercontent.com/uzi18/ESPEasy/6ae4fbc393674fdf1e53546fd5e533e0e84b3f26/src/ESPEasy.ino

copy whole page and replace in ESPEasy.ino

rene9900, 2018-02-15:
im getting all sorts of errors, mine is the R120 i see this ino is for mega, the R120 was the only one i could fit on my esp-01

uzi18, 2018-02-15:
It's a bit ancient.

rene9900, 2018-02-15:
yeah, but it did compile, i updated to 2.3.0 arduino core and tried with mega again, and all is a mess haha

rene9900, 2018-02-16:
i finally succeeded in compiling the mega on core 2.30 with no lib errors "yay"
uploaded to my esp-12 and it works, when i change the ino to the fixed one i do get a odd error 
'thisBinaryMd5' was not declared in this scope
but as the Build | 20100 - Mega (core 2_4_0) reports mqtt vcc as it should im happy, now ill need to see if the -01 will accept the mega, is there no way to run espeasy as minimal ?

TD-er, 2018-02-16:
It looks like you're not cherry picking commits, but just files.
That's always a bit dangerous in the sense that you may run into bugs.

I was told the core 2.3.0 is already so 'big', that you'll have to strip almost all to fit into 512k. Also the SPIFF may take a relatively big part from the 512k.

rene9900, 2018-02-16:
Well thanks for the help all of you, im happy 👍  IDE serial monitor outputs correctly now on my esp-12

INIT : Booting version: (custom)
INIT : Rebooted from deepsleep #2
FS   : Mounting...
FS   : Mount successful, used 76304 bytes of 113201
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:25408
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be 8266-5_22 with address 192.168.4.1
IP   : Static IP :251766976
WIFI : Connecting WIFI attempt #1
Current Time Zone: STD time start: 2018-10-28 03:00:00 offset: 60 min
WIFI : Connected! IP: 192.168.1.15 (8266-5_22)
MQTT : ClientID 8266-522 connected to broker
Subscribed to: /8266-5
EVENT: System#Boot
ACT  : delay 2000
ACT  : Publish /8266-5/status,8266-5_Online
ACT  : gpio,15,1
SW   : GPIO 15 Set to 1
ACT  : Publish /8266-5/Time,09:10:08
ACT  : Publish /8266-5/ip/IP,192.168.1.15
WD   : Uptime 0 ConnectFailures 0 FreeMem 20688
SYS  : 4.47
EVENT: VCC#VCC=4.47
EVENT: Clock#Time=Fri,09:10
EVENT: System#Sleep
ACT  : Publish /8266-5/status,8266-5_Going_to_Sleep
SLEEP: Powering down to deepsleep...

uzi18, 2018-02-16:
Ok so it is with my changes?

rene9900, 2018-02-16:
> Ok I assume you did test with my changes?

> 
i did try the changed ino file, but got the error i posted:
'thisBinaryMd5' was not declared in this scope

uzi18, 2018-02-16:
@rene9900 just buy 32Mbit flash and solder it on esp board.

rene9900, 2018-02-16:
I tested on a wemos mini, the deepsleep 0 which worked on R120 on both wemos and -01 do not work unless a jumper cable connected reset to gpio16, any suggestions? Since my solder skills is too weak to solder a wire to reset -> gpio16 on the -01

TD-er, 2018-02-16:
Like [this](http://domoticx.com/esp8266-wifi-stroomverbruik-en-slaapstand/) you mean?
[image]

uzi18, 2018-02-16:
It is not so difficult as it is on the corner :)
Try to use thin kynar wire it will be more easy.

closed 2018-02-16 by psy0rz

rene9900, 2018-02-17:
> ike this you mean?

> 
yes, like that...i cust cant figure out what the difference is from R120 to the Mega firmware that makes it change so it is needed to solder that wire to GPIO16, i can use deepsleep in the obsolete firmware version with no wire at all?

TD-er, 2018-02-17:
If you can do that without soldering, I am also quite curious how it should be done.

I thought I read somewhere someone made a comment about that, including how to do it, but I couldn't find it yesterday.
Not sure if it was here, or on the forum.

PR #875 (2018-02-16): provide System#Sleep event
- provide System@Sleep event
- fixes #870 (mqtt not sent before sleep)
- call deepsleep at the end of loop - so we know all is done
commit 2b7a6146 (2018-02-16): provide System@Sleep event (#875)

- provide System@Sleep event
- fixes #870 (mqtt not sent before sleep)
- call deepsleep at the end of loop - so we know all is done
PR #931 (2018-02-25): provide System@Sleep event (#875)
- provide System@Sleep event
- fixes #870 (mqtt not sent before sleep)
- call deepsleep at the end of loop - so we know all is done
commit 67c8477a (2018-02-25): provide System@Sleep event (#875) (#931)

- provide System@Sleep event
- fixes #870 (mqtt not sent before sleep)
- call deepsleep at the end of loop - so we know all is done