letscontrolit/ESPEasy #871: DHT12 on I2C send data fixed

tuiting, 2018-02-14:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
DHT12 on I2C send last data after bugging

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Reboot of ESP don't correct the bug. After Power off/on, ESPEasy send correct data.

### Expected behavior
Tell us what should happen?
Temp and Humidity changing by move sensor in different condition.

### Actual behavior
Tell us what happens instead?
Temp and Humidity not changing by move sensor in different condition.

### System configuration

Hardware: Node MCU V3

Software or git version:Release v2.0-20180214

TD-er, 2018-02-14:
So, the sensor keeps sending data like there's nothing wrong, but internal the sensor does not update its values?

Maybe possible to send a reset to the sensor when data has not changed in the last N minutes?

How often does this happen?

tuiting, 2018-02-21:
Thanks, for your reply. Yes, the sensor keeps sending data like there's nothing wrong, but internal the sensor does not update its values.
I have not tested to send a reset to the sensor but I have flashed with "ESP_Easy_v2.0-20180213_normal_ESP8266_4096" normal firmware and since there no issues.

closed 2019-10-11 by Grovkillen