letscontrolit/ESPEasy #873: FS : Error while reading/writing security.dat in 1038

s0170071, 2018-02-14:
No idea how I got there but on boot serial shows
```
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...OK
FS   : Error while reading/writing security.dat in 1089
CRC  : SecuritySettings CRC   ...FAIL

```

When I change devices I get a red warning in the web interface 
FS : Error while reading/writing security.dat in 1038

The info page shows the flash contents: 
[image]

Now, where would I find that format spiffs button ?

uzi18, 2018-02-14:
2 times rules1.dat?? Something is wrong with spiffs.

TD-er, 2018-02-14:
What version/build of ESPeasy was used?

And I guess the format button is not that far ahead in the future ;)

---
Verzonden vanaf laptop

s0170071, 2018-02-14:
I used it with various combinations. Latest was mega with 2.4.0 git. Maybe the double files come from flashing with 3M spiffs and then 1M spiffs option on the same ESP.

TD-er, 2018-02-14:
So the actual issue should be called "request for format SPIFF button"?
Should it also create some initial security file with the WiFi settings?
(if they are stored in SPIFF, not really sure)

And now we're at it, what did you think of the new red error banner on
screen? ;)

---
Verzonden vanaf laptop

s0170071, 2018-02-14:
What about two options. 
a) format and recover the settings. If the settings check out , e.g. the checksum is ok, they could be rewritten after format.
b) just format and write blank settings.

I like the red banner ;-)

Grovkillen, 2018-02-15:
Is this 2.0 or 2.1 level?

TD-er, 2018-02-15:
I guess 2.0, since it will reduce the number of issues and speed-up testing.

psy0rz, 2018-02-15:
There is a reset command via serial that will format but indeed the webgui needs on as well.

s0170071, 2018-02-26:
see PR #954

closed 2018-02-26 by s0170071