letscontrolit/ESPEasy #877: Add VCC enable firmware file to release list

Grovkillen, 2018-02-15:
We have more and more people trying to use the VCC information. This makes bug testing a big sluggish since they are not using official builds. We then don't know how the rest of the code looks like. 

Suggestion : we start providing a VCC enabled release for NORMAL with esp8266 sizes 1M and 4M.

Grovkillen, 2018-02-15:
And while we are at it. Put the bin files in sub-folders (normal, testing, development)

Budman1758, 2018-02-15:
When the VCC feature is enabled it connects internally to the ADC pin on the ESP if I'm not mistaken. If this is true does that make it unusable for external ADC functions?
This may explain why its not enabled by default?
If any of this is true it should be pointed out.
Or am I all wet? (VERY possible)

s0170071, 2018-02-15:
Isn't that vcc feature run-time configurable ?

TD-er, 2018-02-15:
Not at this moment, since all vcc-related code is behind a #define.

TD-er, 2018-02-18:
@s0170071 I think adding the filename would also shorten the time to get some people to give the build version.
Take a look at some topics on the forum and you'll see what I mean.

TD-er, 2018-02-18:
I was also looking into this.
Is there a reason why we do not use the ADC_VCC feature as a setting, which requires a reboot to toggle activation?

Grovkillen, 2018-02-18:
Yes, I agree with @TD-er please add the name 👍

TD-er, 2018-02-18:
Hmmm, read [here](https://www.letscontrolit.com/forum/viewtopic.php?p=5808#p5808) that there are some limitations on how to enable it.
If it has to be done outside any function scope, then it would be hard to make it configurable.

s0170071, 2018-02-18:
Alright then. I'll do it.

TD-er, 2018-02-19:
As discussed on the [forum](https://www.letscontrolit.com/forum/viewtopic.php?p=24862#p24862), must also add the vcc option to the Puya build.

TD-er, 2018-02-20:
Hmm, the platform.ini file is merged, but the _vcc versions of the binaries are not included in the zip.

@psy0rz perhaps something is fixed in the release build script?

Micha-he, 2018-02-21:
Same today. No pre-build Vcc-images.

TD-er, 2018-02-23:
If it is not there, then I guess the issue should be open.
I know @psy0rz is working on it though.

Grovkillen, 2018-02-24:
They are not defined in "before_deploy".

psy0rz, 2018-02-25:
yeah it shouldn't been closed. ill update the deploy script for both branches and create subfolders.

Grovkillen, 2018-02-27:
this is now live

closed 2018-02-27 by Grovkillen