letscontrolit/ESPEasy #878: Split command handling + allow to generate list of commands

TD-er, 2018-02-15:
In the command.ino file, there is a long-long-loooooong list of if-the-else statements to filter the commands.

There should be a single function to parse all command names and convert them to an enum.
And the other way around (can be done using a map or something like that) to generate a list of commands as help function on the webpage and via serial terminal.

Then handling the commands is just a single switch statement, which is a lot easier to maintain, to read, etc. Also may benefit memory use, since the lookup could be stored almost entirely in flash.

Another benefit is that there is someone looking through these commands and may optimize some parts, check documentation, etc.

Grovkillen, 2018-02-15:
Great idea, would love to see it implemented like that.

TD-er, 2019-10-27:
This issue is no longer relevant, since we now have a rather efficient way of parsing commands.

closed 2019-10-27 by TD-er