letscontrolit/ESPEasy #880: Framed Display and Software Serial

s0170071, 2018-02-15:
I was applying bursts of 500 bytes @9600 bps to the software serial. 
This effectively stalled the framed display. Scrolling was very slow.

mega / 2.4.0

9600, 2018-02-15:
You called?

s0170071, 2018-02-15:
Ah. no. sorry to wake you. Didn't know you existed actually. I change my serial speed then, ok?

9600, 2018-02-15:
I don't mind what speed you use, although I would say that 9600 bps is a particularly pleasing rate.

I may be biased.

TD-er, 2018-02-15:
No surprise it will stall the ESP.
Although I agree with @9600 about the pleasing rate. Some may call it even nostalgic.
It still is a lot of interrupts over a long period.
That will take at least half a second and probably longer to send 500 Bytes.
During these close to 10'000 interrupts per second, a lot of "delay()" and "yield()" calls are made to keep up with WiFi connections etc.
So there is not much time left to do something else.

s0170071, 2018-02-20:
Just took a look at the ESPsoftwareserial code; it does call the ISR when the start bit is received an then it waits until all bits have arrived. 
How about a re-entrant ISR that assembles the byte from the timestamps of its invocation ? This may work better for low (pleasing) baud rates where it would spend most of the time waiting otherwise...

TD-er, 2018-02-20:
Hmm, I'm sure I already replied on this one.... ???
There is a similar approach in the original library for low-speed transfers, but that resulted in lots of reboots and crashes. So apparently that's not so simple.

TD-er, 2018-03-09:
Can this one be closed?

closed 2019-10-27 by TD-er