letscontrolit/ESPEasy #882: Make user aware that some plugins only work on certain task numbers

tedenda, 2018-02-15:
### Steps to reproduce
How can we trigger this problem?
Go to devices, go to next page (task 5-8), add new device, select pulse counter.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
In my opinion you should be informed in some way that this is not a valid choice because Pulse Counter only works on task 1-4 right now. It could be done by graying out and make it un-selectible in the device list drop down.
 
### Actual behavior
Tell us what happens instead?
You are allowed to configure this device as normal and think that it will work fine. But you will never get it to work because it is disabled in firmware. After some struggle you hopefully hit the help button and see on the bottom of that Wiki-page that you only can place that device on task 1-4.

Software or git version:
Nightly v2.0

TD-er, 2018-02-15:
Also good to check:

- Do not use same name of tasks.
- When changing a name which is used in some rule, add a warning.
- Add a warning when a hostname cannot be resolved.
- Add a warning when a device cannot be detected in a plugin/task  (e.g. I2C devices)

Grovkillen, 2018-02-16:
Length of task name, value name should also be included in the scope. Also it's not recommended to have the same name for task and value. Spaces is names are also a no-go.

s0170071, 2018-02-21:
@tedenda: When adding a task is not working- what does the FreeMem line on the main webpage read ? 
e.g. 

`Free Mem: | 18544 (13888 - duringDataTX)`

TD-er, 2018-02-26:
Do not allow characters other than numbers for the unit number.
See [forum](http://www.letscontrolit.com/forum/viewtopic.php?p=25125#p25125)

TD-er, 2018-02-27:
Show a warning when performing OTA on a 1MB flash device. 
Suggested [here](https://github.com/letscontrolit/ESPEasy/issues/708#issuecomment-369023586).

TD-er, 2018-03-02:
Set msg delay to 100 ms when using a switch.
See [forum](https://www.letscontrolit.com/forum/viewtopic.php?p=25360#p25360)

tedenda, 2022-12-24:
Can I close this issue?
Last edited 5y ago.

tonhuisman, 2022-12-24:
> Can I close this issue? Last edited 5y ago.

Yes, as many configuration checks (and other improvements) have been implemented.

closed 2022-12-24 by tedenda