letscontrolit/ESPEasy #884: strange behaviour with password

rhodopsik, 2018-02-16:
I ran into an strange behaviour, that I can't understand.

I set the password in the web frontend with my computer, after that I'm not able to access the web-page, there is only a white page without a login field. I restarted the esp12e several times, no change.
Other browser at the same computer, no change.

Than, I tried to reach the esp32e by my smartphone -> the login-page is working -> after loggin in with the smartphone, the page at my computer appears!

I reproduced this 3 times. Every time I set the admin password, accessing the login page is only possible with firefox at my smartphone, but after loggin in at this device, the access with the PC is possible. 
I seems like there is an issue not only with handling.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
-->> YES

Hardware:
ESP12E

Software or git version:
ESP_Easy_v2.0-20180214_normal_ESP8266_4096.bin

mattlward, 2018-02-19:
I am having that problem with Bitdefender, had to add an exclusion.  Drove me nuts, did not see anything logged in Bitdefender, page just never loaded.

TD-er, 2018-03-09:
Can this one be closed?

mattlward, 2018-03-09:
I think so...

closed 2018-03-09 by TD-er