letscontrolit/ESPEasy #885: Add the MAC adress as a system variable

goldriver2, 2018-02-16:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

TD-er, 2018-02-16:
Ah good one. Quite easy to do also. :)

Grovkillen, 2018-02-17:
Would it be possible to convert it as integer as well?

TD-er, 2018-02-17:
It can, but then I have to use 64 bit int.
Mac is 48 bit.
But why would you want that?

Grovkillen, 2018-02-17:
Might be used in rules.

Grovkillen, 2018-02-17:
Skipping the OUI part?

TD-er, 2018-02-17:
I can create some 31 bit value which consists of a 7 bit "checksum" of the OUI part and the 24 bit of the rest.
Better not use it as a 32 bit integer, since that will lead to issues with "<" and ">".

Grovkillen, 2018-02-17:
Sounds like a good idea.

TD-er, 2018-02-17:
The standard calls to Arduino core only return the last 3 bytes (24 bits) and that's also what is being shown in the info page.
So I'll leave the integer part to 24 bits.

TD-er, 2018-02-17:
Should be there in the next build.

Grovkillen, 2018-02-17:
Nice 😎

closed 2018-02-26 by psy0rz