letscontrolit/ESPEasy #890: [BMP280] Fix regression on #636

TD-er, 2018-02-17:
See topic: http://www.letscontrolit.com/forum/viewtopic.php?f=18&t=4864
Was fixed in #636 

Bug was re-introduced in: https://github.com/letscontrolit/ESPEasy/commit/3922827fad617c37f0ee4f4b1d3d5dfb5ca4838e

So @psy0rz if there was a good reason for this change, please have a look at it.
Or else, I will create a PR to fix it again.

TD-er, 2018-02-17:
Hmmm I think there is just too much overlap between C001 and C002.
The differences are probably bugs fixed in one and not the other.

The last sentence of this paragraph suggests the order of the values should be the same for HTTP and MQTT: https://www.domoticz.com/wiki/MQTT#Domoticz_to_MQTT
And there are more examples in which they are not the same.

TD-er, 2018-02-17:
Added a pull request #891 
Please test these with as many sensors as possible.
At least it should not matter anymore if HTTP or MQTT Domoticz controller is used.

psy0rz, 2018-02-20:
yeah there where indeed some strange differences. ill run the controller tests now, it will create dummy for all datatypes, so it will fully test domoticz http and mqtt.

psy0rz, 2018-02-20:
you either fixed or broke SENSOR_TYPE_TEMP_BARO.

psy0rz, 2018-02-20:
INFO:esptest:*** All tests completed ***

closed 2018-02-20 by psy0rz