letscontrolit/ESPEasy #898: light sleep mode

crispyduck00, 2018-02-19:
Is it possible to implement light sleep mode in combination with a delay (e.g. 250ms) in the loop on ESP Easy to reduce power consumption?

When for example the ESP is only used for switching something or while it is in standby mode till its activated a small configurable delay  of 0 to 300ms should not be a problem.

This is also implemented in the Tasmota FW for Sonoff devices, and tests show me that it lets reduce the power consumption when the ESP is in "standby" up to 50%, while it still reachable via wifi.

TD-er, 2018-02-19:
You mean this setting, described on the [Tasmota wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Energy-Saving)?

crispyduck00, 2018-02-19:
Yes, there it is just called sleep; behind is the ESP light sleep mode combined with a delay in the loop.
[https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/272](url)

TD-er, 2018-02-19:
Sounds like a great feature.
@Grovkillen may decide if it will be a 2.0 or a 2.1 feature :)

Maybe we can make it somewhat dynamic, or at least add a warning when people want to use it combined with interrupt driven plugins.

crispyduck00, 2018-02-19:
Great and nice to hear that you also think this will be a good feature for ESPEasy. :)

Grovkillen, 2018-02-20:
If it's fairly straight forward to implement I'd say it's 2.0 👍

Grovkillen, 2018-05-31:
We ditched the 2.0 release since the RC branch got stale. So this is the due for 2.1 or what do you say @TD-er ?

pfeupfeu, 2018-11-03:
Light sleep would be a great feature. I hope you can make it happen.

crispyduck00, 2019-04-25:
Any update on this?

TD-er, 2019-04-25:
[image]
Tools => Advanced => Eco mode

See also the [documentation](https://espeasy.readthedocs.io/en/latest/Tools/Tools.html?highlight=eco)

closed 2019-04-26 by crispyduck00