letscontrolit/ESPEasy #899: DHT22 readings

litrinhos, 2018-02-19:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Just connect a DHT22 sensor properly configured to read periodically (60 seconds).

Does the problem persist after powering off and on? (just resetting isnt enough sometimes)
Yes it persist.

### Expected behavior

The sensor should output valid measures each reading.

### Actual behavior

Only one out of five measures is correct. The other four out of five return "NO READING"

### System configuration

Hardware: ESP12E + DHT22

Software or git version: Release v2.0-20180214

Could be related with:
https://github.com/adafruit/DHT-sensor-library/issues/48

After downgrading to R120 it works seamless. All readings are valid.

TD-er, 2018-02-19:
Well, I think we could try to comment out 2 lines, as suggested in the link you gave :)

TD-er, 2018-02-19:
Could you test this fix?
Either by compiling yourself, or wait until it is merged.

TD-er, 2018-02-19:
Hmm in the Mega branch this was already fixed:

https://github.com/letscontrolit/ESPEasy/blob/3032463917477352da9b3e01d4bd25a52d3408f6/src/_P005_DHT.ino#L76-L82

litrinhos, 2018-02-20:
Hi, I've tested the fix, merged in Release v2.0-20180220.
Same result, only some valid measure from time to time, the rest return "No reading!".

TD-er, 2018-02-20:
Could you also test the Mega version?
If that's working fine, I will backport those fixes into the v2.0 branch

litrinhos, 2018-02-20:
Hi, tested the Mega version. Release mega-20180220.
It works !!! 

I'm reading DHT22 every 60 seconds. I'll try now with deep sleep and 300 seconds between measures, which are the settings I actually need. 

But I'm a bit confused, I know Mega is development branch, but if the stable branch is still under development, why maintaining both branches for developing?

Thanks for your feedback.

TD-er, 2018-02-20:
The 'stable' v2.0 branch is not yet released. 
The current status of the v2.0 branch is feature freeze and fix bugs to make it actually stable.
And in fixing bugs, some (very basic) new features are introduced to keep merging a bit easier.

I will merge the Mega version of this plugin to the 'stable' branch.
Thanks for testing.

litrinhos, 2018-02-20:
Ok, thanks for clarifying.

psy0rz, 2018-02-26:
should be fixed, please reopen if this problem still exists.

closed 2018-02-26 by psy0rz

JoostDkr, 2022-02-15:
It looks like the error is still there in ESPEasy_ESP32_mega-20211224.
Regularly the log shows "DHT: readings not received".

The fix was to modify _P005_DHT.ino line 163 (case P005_DHT22: delay(1); break;) in "case P005_DHT22: delay(10); break;"
:-)

TD-er, 2022-02-15:
I think we may need to make this delay a configurable variable, as it has been changed a lot, based on what users reported.