letscontrolit/ESPEasy #900: Generic HTTP Controller publish URL too short

laacz, 2018-02-19:
### Steps to reproduce
Open up your EspEasy Mega interface and go to Controllers.
Problem persists, of course.

### Expected behavior
I should be able to enter URI at least 254 chars long. I wouldn't call it a bug, if there a controller type "Generic HTTP Advanced", mentioned in manual. Yet, there is only "Generic HTTP".

### Actual behavior
String `submit.php?name=%sysname%&task=%tskname%&valuename=%valname%&value=%value%&vcc=%vcc%&ip=%ip%&uptime=%uptime%&tskname=%tskname%&id=%id%&vname1=%vname1%&vname2=%vname2%&vname3=%vname3%&vname4=%vname4%&` gets trimmed to 128 characters (`submit.php?name=%sysname%&task=%tskname%&valuename=%valname%&value=%value%&vcc=%vcc%&ip=%ip%&uptime=%uptime%&tskname=%tskname%&i`).

### System configuration

Hardware: Wemos D1 mini pro
Software or git version: v2.0-20180218

TD-er, 2018-02-19:
The advanced version is in the Mega branch.
Not sure if that one allows longer URLs, since it is probably limited by the storage layout.

TD-er, 2018-02-19:
They are currently almost the same, with a few enhancements. 
The main difference is in the plugins, but there have not been much changes in them lately.

N.B. The v2.0 is still in dev phase and we are working very hard to release it.

laacz, 2018-02-19:
OK, Tnx.

laacz, 2018-02-19:
I'm sorry I have to do this, but there is still no advanced HTTP in mega version (mega-20180219):

[image]

TD-er, 2018-02-19:
I'm sorry, I just looked into the code: https://github.com/letscontrolit/ESPEasy/blob/v2.0/src/_C011.ino
it is even in the v2.0 branch, but it is still "testing", so you have to get the "testing" build, or set the testing flag if you build it yourself.

sassod, 2018-02-19:
Its in the testing and dev firmware.bin or if you now how build it yourself
build the test or dev build.

laacz, 2018-02-19:
OK. Closing then.

laacz, 2018-02-19:
P.S. Probably should have been mentioned in manual :)

closed 2018-02-19 by laacz