letscontrolit/ESPEasy #901: System variable assigned to Git version?

mattlward, 2018-02-19:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
It would be great if we could do %version% and then use that variable as an mqtt published value.

Not one listed here:
https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:
Wemos
Software or git version:
v2.0-20180219

TD-er, 2018-02-19:
Do you have an example output from the JSON Tasmota will generate?

And do you have a suggestion on the Git version? The SHA will not be very useful for many.
I also experimented myself with branch names. That was for my own builds, but it will not work for the release builds, since these are always made in either 'mega' or 'v2.0' and there is already an issue about including the build filename.

mattlward, 2018-02-19:
Here is one of the outputs.  It is like a dictionary with with key:record pairs that can be picked out in code.
[image]

On the main screen of the ESP there is a field called "GIT Version" with v2.0-20180219.

That could be used to determine quickly the version running in Home Assistant and could indicate if an update is needed.

TD-er, 2018-02-19:
Json output is best served as text, not in a screenshot ;)
That's easier for development

mattlward, 2018-02-19:
Sorry, here are a couple of example strings:
`
{"Time":"2018-02-19T15:37:16","ENERGY":{"Total":161.392,"Yesterday":0.511,"Today":0.006,"Period":0,"Power":0,"Factor":0.00,"Voltage":119,"Current":0.000}}
`
`{"Module":"Sonoff Pow","Version":"5.12.0","FallbackTopic":"DVES_3B7652","GroupTopic":"sonoffs"}
`

TD-er, 2018-02-19:
Related:
#862
#730

micropet, 2018-02-21:
The Main screen always shows:

GIT version: | (custom)

Grovkillen, 2018-10-23:
We should add this

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

TD-er, 2023-09-04:
Yep, we do have `%sysbuild_git%`, `%sysbuild_filename%` and `%sysbuild_desc%`, which can be used in any command accepting string values (e.g. MQTT publish)

closed 2023-09-04 by TD-er