letscontrolit/ESPEasy #907: UDP Payload

dtavares07, 2018-02-20:
Hi.
It's possible to send a command to the ESPEasy via UDP from an arduino?

Best regards,
Daniel Tavares

uzi18, 2018-02-20:
This command is to send something from espeasy by udp.

uzi18, 2018-02-20:
You need only to send packet to udp port with command to execute. 
Udp port is configured in advenced options.

dtavares07, 2018-02-20:
THX for your reply.

And what should be the syntax of that command??? because I can't figure it by the documentation provided.

uzi18, 2018-02-20:
Use UDPSendReceiveString.ino example. 
You need only to use these functions:

  Udp.beginPacket("192.168.4.1", 125);
    Udp.write("gpio,2,1");
    Udp.endPacket();

Change ip and port 125 fot your espeasy configuration.

dtavares07, 2018-02-20:
Thanks I'll try it :)

uzi18, 2018-02-20:
Commands are all from command reference look above link from Matt

uzi18, 2018-02-20:
Give us some info if it work. 

@Grovkillen @TD-er,  do we need also udp firewall?

dtavares07, 2018-02-20:
It works. :)

Thx guys

TD-er, 2018-02-20:
I haven't looked into what UDP things we currently support.
As far as I know, it is "just" for communicating values and not for sending commands and changing settings.
But if that's not correct, then I guess we may need some simple filtering.
Maybe it is good to make a list first of all things supported:

Commands via: HTTP/MQTT/serial/whatelse?
Change settings via: HTTP/serial/???
Read values via: ???
Trigger rules via: GlobalSync, HTTP, MQTT, ????

And what else could be used as an attack vector?

uzi18, 2018-02-20:
As You can see it is possible to execute command anonymously, by udp packet.

dtavares07, 2018-02-20:
Just to see if I've understood. What I've asked is a bad thing?

uzi18, 2018-02-20:
No it is ok.

mattlward, 2018-02-20:
Talking about the command reference... all of the stuff below that is not nicely formated... is there any expansion elsewhere that goes into more detail?

TD-er, 2018-02-20:
@dtavares07  "finding" and reporting on things that allow for someone to do bad things is never a bad thing.
The 2.1 release should be more about security, that's for sure ;)

TD-er, 2019-11-04:
Improving p2p is on the radar and since this issue does not really describe what has to be done, I will close this one.

closed 2019-11-04 by TD-er