letscontrolit/ESPEasy #909: Rules: conditions works wrong

uzi18, 2018-02-20:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce

```
on Rules#Timer=1 do
 if [something#value] > 14
   if [other#value] = 1
     command1
   endif
 else
   command2 
 endif
endon

```
### Expected behavior
if  [something#value] > 14  and   if [other#value] = 1 are true, rules must execute only command1

### Actual behavior
if  [something#value] > 14  and   if [other#value] = 1 are true, rules execute command1 and then command2

### System configuration

current mega 20180220 release.

sassod, 2018-02-20:
From wiki

only simple if/else is possible - so nesting and Boolean logic are not
supported. However there is a workaround for the limitation of not being
able to nest. An "event" can be called from a "trigger".

on <trigger> do
 if <test1>
   event <EventName1>
 endifendon

on <EventName1> do
 if <test2>
   <action>
 endifendon

uzi18, 2018-02-20:
@sassod  know about it but it is not a solution, as it delay rules execution and is not readable.
I will try to fix this

TD-er, 2018-02-27:
This one is fixed now, right?

uzi18, 2018-02-27:
Yes,  also small fix is needed to and/or conditions,  wait for PR.

TD-er, 2018-03-09:
Can this one be closed?

uzi18, 2018-03-09:
It is fixed right now

closed 2018-03-09 by TD-er