letscontrolit/ESPEasy #911: Feature: issue html page + button.

s0170071, 2018-02-20:
I thinḱ it would be handy to have an "i found an issue" button on the ESP web interface that directly opens an issue at  github. That button also opens a html page on the ESP that includes all relevant information like build, branch, hash, .... ready to copy and paste into the github issue.  

Bonus: if already logged in, that info text is passed as url parameter (e.g. github.com/issue.htm&mytext=...)  to github, if that is possible.

TD-er, 2018-02-20:
Not sure how you would want to implement that, but I like the idea.
But generating something that can be cut'n-paste would already be an improvement.

mattlward, 2018-02-20:
That is a great idea!

Grovkillen, 2018-02-24:
Great idea, we'll need to think on how to go about implementing that but that's for later.

s0170071, 2018-02-24:
I checked with github: 

> You can use the body parameter. For example:
> https://github.com/letscontrolit/ESPEasy/issues/new?body=somerrordescriptionAndSystemInformation
> 
> You can also set the title:
> https://github.com/letscontrolit/ESPEasy/issues/new?body=somerrordescriptionAndSystemInformation&title=Title+Goes+Here
> 
> As well as labels and other settings. You can read more in this article:
> https://help.github.com/articles/about-automation-for-issues-and-pull-requests-with-query-parameters/

I would contribute that function but would wait until the 2kwebserver #929 is merged.

TD-er, 2019-10-27:
We have the copy for Github to help people create a new issue with the basics of information.
Also we have a template, which is ignored in about 80+% of the newly opened issues.

Not sure if this is still a good idea to implement.
The current tools are not being used and if we put in a lot of effort to only see it not being used and we still have to ask for it, then I'm afraid we either get lots of similar reports or keep asking for "press this button please".

So for now, I will close the issue. Please comment if you still think it is a good addition.

closed 2019-10-27 by TD-er