letscontrolit/ESPEasy #916: Is it possible to get SNMP V1 or V2c?

mattlward, 2018-02-21:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Not installed...
Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?
Unit should answer SNMP queries
### Actual behavior
Tell us what happens instead?
nothing
### System configuration

Hardware:
Wemos D1
Software or git version:

TD-er, 2018-02-21:
Hmmm the ESP should be powerfull enough, but I am not such a masochist to implement it. ;)
If there is someone who will write a very simple implementation for it, be my guest. A good example can be [Agentuino](https://github.com/1sw/Agentuino).

I'm afraid implementing it would be only a fraction of the time needed to make it work like everyone would want it to work.
Apart from that, allowing to set settings via SNMP would be a big security issue, so preferably only reading.

mattlward, 2018-02-21:
I would love to be able to ask for sensor data...  I use intermapper to montor about 2500 devices and there is no good way to get mqtt data into it.

TD-er, 2018-02-21:
What do you want to read from it?
Only the device status, or also sensor data?
And how should it be organized in SNMP?

---
Verzonden vanaf laptop

mattlward, 2018-02-21:
I would want to read sensor data... I would expect a tree under the private mib structure with a branch for each sensor type.  

private ------>
    sys-------->
        sht30--->
        dht22--->
        so on--->

all would be read only...  

I have had no luck finding an MQTT to SNMP gateway, this is the problem with IOT devices, they are just not designed to talk with the professional world.

TD-er, 2018-02-21:
Are there other xxxx-to-SNMP gateways?
I am planning to implement a REST api, which is perhaps a bit more
future-proof?

I'm all in for interfacing with the professional world, but implementing a
protocol from the 80's, or the v2 from 1993, where there is no sense of
security, feels a bit "mwah".
And if I remember correctly, the v3.x is not that common. Also SSH/TLS is
currently a bit of an issue while still using core 2.3.0.

I know in the industrial world some things just never stop being used and
things tend to remain a bit 'traditional'.

What datastypes should be supported?

   - Get
   - GetNext
   - Set => no way!
   - GetBulk
   - Trap => would involve a lot of changes in ESPeasy settings

---
Verzonden vanaf laptop

mattlward, 2018-02-21:
I would forsee using get and get bulk, get bulk is cpu intensive.  I come from the world of a very large network... we have 2500+ very modern switches, routers, air handlers and ups units that all use SNMP.  We use V2c when possible since it supports all forms of request and transmission.  Would never suggest using sets in this environment.  

Traditional is all the modbus and RS485 running around in modern building system controls!

A rest API might work, but would require more complex processes when a call is made to the device?

The ideal thing would be a Pi running as an MQTT client with an SNMP frontend that had a definable MIB that could be tied to the MQTT topics.  Can't find one!

mattlward, 2018-02-21:
BTW, misspoke... should not be in the private mib structure but rather the experimental mib structure.

clumsy-stefan, 2021-08-31:
Just came across this while setting up a new NMS... Are there any thoughts on pursuing this any further? There is an updated SNMP library at https://github.com/rexpark/Arduino-SNMP not sure though how complex this would be to integrate...

TD-er, 2021-08-31:
I do use LibreNMS for myself and to monitor a larger network of a client of mine.
So now that I have seen the practical usage of SNMP, I do plan on adding it.

N.B. LibreNMS is a truly marvel and has given me a lot of insights and saved me a lot of research to see what may be causing an issue.

clumsy-stefan, 2021-08-31:
Same here... LibreNMS is watching my network... which gives me a good overview of some 100 IoT and other IP Systems... so it would be kind of cool to have more information on ESPEasy nodes than jsut an ICMP reply...

TD-er, 2021-09-01:
Not 100% sure if it should be a controller.
Typically a controller does send data and SNMP is polling.
Unless we also implement the trapping functionality of course.

So maybe a controller is the best approach, but I'm not yet entirely sure.

The reason is that a controller is usually not aware of what it is sending, but for SNMP you need to know what you try to poll.
Also we need to add an MIB describing the paths to data.
But since we don't know now what we will support in the future, this may be a lot of work to keep it up to date when we now make the wrong design choice.
So maybe such a controller should have a number of predefined "paths" which a user can select using an IDX value.

I don't know if SNMP also supports to automatically fetch an MIB file, or something similar. But if it does, we may also need to implement a hook to supply a dynamically generated one.

Just to begin, we can of course support the basic system parameters, like uptime, load, memory, WiFi signal strength, etc.
But those are system functions and thus do not really need a controller. As soon as you want to report more, it makes sense to make it a controller.

clumsy-stefan, 2021-09-01:
Sounds like a reasonable approach to me... 

I would not implement traps at this time, just responses to queries (which also a controller does?) and as you said, just basic standard MIB information like the ones you mentioned. As it's normally UDP, the controller doesn't really need to know if it's been received, etc....

I'm not completely sure, but I think you don't need the MIB file but just send the correct OID's in the response.

TD-er, 2021-09-01:
Well being polled instead of sending means that the controller must keep all values, just like a retain flag with MQTT.
Also it must support the "SNMP walk" and I am not sure what is needed to support that.
Have to dive into the SNMP specification a bit more.

I think we should only support SNMP v2.x as v1 is too basic and v3 does require authentication.

clumsy-stefan, 2021-09-01:
> Well being polled instead of sending means that the controller must keep all values, just like a retain flag with MQTT.

Hmm.. can't it just reply with actual values when queried?

Other possibility would be to have something like the /json page. Just sending actual basic values in a well predefined format, without any/much config possibilities. But I'm also not an SNMP specialist.. 

> I think we should only support SNMP v2.x as v1 is too basic and v3 does require authentication.

Agree on that, just reply to v2c read communities (probably with a configurable community string as basic security)

clumsy-stefan, 2023-01-29:
A quick update on this. I am currently switching to zabbix as new NMS. As zabbix can easily integrate JSON information I developed a zabbix template to monitor my ESPEasy's. Currently only on a system level, i.e. without sensor data, which I use and monitor in FHEM.

So if anyone is interested, I'm happy to share my (work in progress) template!

TD-er, 2023-01-29:
OK, while you're using Zabbix, can you perhaps also have a look at [this forum topic](https://www.letscontrolit.com/forum/viewtopic.php?p=62514&hilit=zabbix#p62514)?

clumsy-stefan, 2023-01-29:
Done, see https://www.letscontrolit.com/forum/viewtopic.php?p=62804#p62804

TD-er, 2023-01-29:
I meant there is some user claiming to not being able to use the Zabbix controller at all on ESP32, so maybe you can see where the issue he's having may be coming from?

clumsy-stefan, 2023-01-29:
I don't use the Zabbix controller at all! I use active polling from zabbix, not the other way round, my ESPEasy's dont have the zabbix controller compiled in, as I keep it as small as possible and don't use zabbix for the sensor values...