letscontrolit/ESPEasy #919: Mega - massive memory loss

s0170071, 2018-02-21:
2.4.0 / mega

With no devices active, no logging memory is 

```
Free Mem: | 19600 (8864 - sendWebPageChunkedData)
-- | --
```

after clicking on each main menu once.
After enabling the framed display it goes

```
Free Mem: | 17224 (4672 - sendWebPageChunkedData)
-- | --

```

Memory has been low before but thats new.

TD-er, 2018-03-07:
Have you seen my latest patch on your streaming class? #1027
I've already merged it.

Now it is really lightning fast. Loading the OLED Framed page on my Chromebook (which has a Celeron inside) takes about 420 msec. The Advanced view page takes 438 msec.
And memory use stays within 1k.  (tested on core 2.3.0)

s0170071, 2018-03-07:
This issue was pre-webserver2k ;-)
Just looked at it.  As I understand it you did two things: 
1) sending the css not byte by byte to the txBuffer but in chunks that end with "}" or ";"
2) adding the string thats sent to the TXBuffer not as one but byte by byte.

As for 
1) I don't understand why.  
2) bings the benefit of a little less ram used *if* you would have gone beyond the reserved size of the TXbuffer. The memory required for "a" is still used.

What exactly is the speed benefit of these changes and where are they coming from ?
What about the memory drop on display "submit" ?

TD-er, 2018-03-07:
1) The original code did create a String object per byte. now it is re-using it with a String with reserved allocated space. The separation per } or ; was just because I did not know whether there would be some other chars inserted when sending chunked. That was my first attempt to speed up the code and may be room for some improvement.

Then I thought about how to do the same for the other larger strings, which brings us to 2) ;)
The problem with the original implementation of operator+= const String& was that it could go way beyond the buffersize of 400. There are some strings which are much longer than 400 bytes (e.g. css/templates) and these need to allocate more room for the buf in the TXbuffer object, which probably will not be released for a while.
The chunks will now be as close to 400 bytes as possible when sending large strings.
I've seen a lot of packages with > 400 bytes but not > 450 bytes, when looking at it with Wireshark, so it must be doing something right :)
My trigger to look at it was a few "non responsive" and even "hang for longer time" actions after I updated the CSS, which is about 3.4k

There is still some room for improvement by using strcpy instead of doing it one byte at a time. Also for dumping flash stored strings, can use the read_word functions and the strcpy_P functions, which will read 4 bytes at a time.
But it was already past 1am and using such functions require some 32-bit memory alignment, flushing the last 3, 2 or 1 byte, etc.

I did not test submit yet, only looked at the outgoing part.

s0170071, 2018-03-07:
Why do you wait for an } or ; ? You could just fill up that 64 byte temp string unti its full...

TD-er, 2018-03-23:
I guess this one can be closed.
If not, please re-open.

closed 2018-03-23 by TD-er