letscontrolit/ESPEasy #920: _N001_Email & #define FEATURE_ADC_VCC true compile error : const NotificationSettingsStruct& notificationsettings not defined

mvroosmalen1970, 2018-02-21:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
compile with #define FEATURE_ADC_VCC true with build 20180221 and later and some earlier versions.

this reports:
_N001_Email:67: error: 'NotificationSettingsStruct' does not name a type

 boolean NPlugin_001_send(const NotificationSettingsStruct& notificationsettings, const String& aSub, const String& aMesg) {

                                ^

_N001_Email:67: error: ISO C++ forbids declaration of 'notificationsettings' with no type [-fpermissive]

 boolean NPlugin_001_send(const NotificationSettingsStruct& notificationsettings, const String& aSub, const String& aMesg) {

                                                            ^

F:\arduino-1.6.12P230\Portable\sketchbook\ESPEasy\_N001_Email.ino: In function 'boolean NPlugin_001(byte, EventStruct*, String&)':

_N001_Email:59: error: invalid initialization of reference of type 'const int&' from expression of type 'NotificationSettingsStruct'

         NPlugin_001_send(NotificationSettings, subject, body);

                                                             ^

_N001_Email:67: error: in passing argument 1 of 'boolean NPlugin_001_send(const int&, const String&, const String&)'

 boolean NPlugin_001_send(const NotificationSettingsStruct& notificationsettings, const String& aSub, const String& aMesg) {

         ^
after removal of _N001_Email. The build is compiled with only a few lib warnings.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
see above
### Actual behavior
Tell us what happens instead?

### System configuration
Windows10
Hardware:

Software or git version:
arduino-1.6.12P230 fresh install with build 20180221 of esp easy.

TD-er, 2018-02-21:
There is a forward declaration of that function definition in ESPeasy.ino
Move that to a point above where the first define of this VCC is being used.

But I thought I already fixed that very recently.
Will look at it tomorrow

TD-er, 2019-10-27:
Building is no longer an issue for this. So issue can be closed.

closed 2019-10-27 by TD-er