letscontrolit/ESPEasy #922: Increase retry interval when controller cannot be reached

TD-er, 2018-02-21:
When using sensor in another network, WiFi is enabled, but e.g. MQTT broker cannot be reached.
Then the load will be quite high and ESPeasy rather unresponsive.

Seen with domoticz MQTT enabled.

mattlward, 2018-02-22:
I see the same issue with a unit on a remote network connecting to my HA server at home.  Retries are very fast, sometimes it will show a quick connect and the disconnect for several attempts before settling in.  I am using Mosquito on  Home Assistant with Hassio.

TD-er, 2018-02-22:
N.B. this also is applicable for NTP and DHCP requests + retries.

uzi18, 2018-02-23:
I have also problems with mqtt connection, it connect/disconnect too fast
so connection is not possible.
Log in mosquito is like: host connected, host disconnected ;)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Grovkillen, 2018-03-16:
I must say that this is fixed... at least for my units. I will reopen if OP disagree.

closed 2018-03-16 by Grovkillen

mattlward, 2018-03-16:
Not sure... my broker was down 2 nights ago and with serial connected to a Wemos I could see the broker retries but could never get a connection up to the units ip address.  It only became responsive once the broker was back online.  

Not sure of the version, but I am never more than a day or 2 behind on the V2.0 nightly builds.