letscontrolit/ESPEasy #923: emergency action on low memory

s0170071, 2018-02-22:
In order to be able to detect that an issue is low memory related I propose the following: 

In the **sendHeaderblocking()** function (pseudocode):
```
 if freeHeap() < 3500 {  // value depends on build and lwip.
   send(200,"OK","low memory");  // just a quick response, nothing fancy
   wait for core memory to return.
   leaveNoteOnSpiffs()
   ESP.restart();
}
```

 
In **checkMem()** : 
```
 if freeHeap() < 3500 {
    leaveNoteOnSpiffs();
   LOG(low memory);
   ESP.restart();
}
```

in **setup()**

```
check if there is a low mem file on spiffs
if yes:  {
   disable all plugins 
   display a warning on Webpage 
   display warning on LOG
}
```

@TD-er @Grovkillen  @psy0rz  and all others: What do you think ? 
you like it- i'ld volunteer. Those memory issues begin to cause a lot of work and confusion.

Budman1758, 2018-02-22:
Just curious...  What is considered low?  And are we talking about the Free Mem that's listed on the main page?

Free Mem: | 14952 (5472 - sendWebPageChunkedData)

TD-er, 2018-02-22:
@Budman1758 It's about the second number in your quote.
The lowest free memory since last check.

I think writing to flash may also use quite some memory, so I am not sure that will be the best way.
There is 1 byte stored in memory to report the boot cause. Maybe there is some room to set an OOM (out-of-memory) value there?
That should be discussed with the guys from Arduino ESP8266 Core (perhaps @igrr ??) to make sure no clashes with future added values will occur.

But I like the idea.

Perhaps not all should be disabled, but not sure how to detect what caused the memory issues. (and store that in 1 byte)

s0170071, 2018-02-22:
I found that the figure in the braces (indicates the lowest memory during user program execution) should be larger than 3000. Depending on what wifi packets are incoming at the same time - some more. I would go for 5000bytes min free. Just to be safe. 
I rather have the program stopped like this than having undefined and mysterious behaviour. Those are bugs that are hard to find without debugger.

s0170071, 2018-02-22:
@TD-er the amount of mem required writing to flash can be found. I can give it a try. For example:  I just block all memory except for 1k and write a file. If I don't build large strings and just create a file (with one character in it) that might not be that much. 
Writing spiffs should be working at all times anyway, so we need set the memory threshold so that it does. 
As for disabling the Tasks: Unless there is a way to tell which one consumes which amount of memory or which one causes the memory problems (leak, large strings etc.) I don't see any reliable way other than disabling them all. Only the user knows what he did last. 
Note, I don't mean deleting the config, just disabling it so its not instantiated on next boot.

TD-er, 2018-02-22:
That last sentence may need some verification in the code. ;)
Not all plugins/controllers, etc do nothing when disabled. I fixed that in a few cases already.

And writing to SPIFF should be constant in memory-use, I agree.
There should be an option to temporary block incoming traffic. I've seen the webinterface hang for a while when you try to access it too soon after reboot (Chrome will retry a lot on its own)

Another thing during reboot, we should start all things one after the other, not all at once.
I want to change the way how certain tasks are being dealt with and give each their own scheduling timer and split tasks in subtasks to speed things up and probably reduce memory use.
For example retry to regain connection should be a stage and not part of a big one-call-does-it-all process.
And perhaps there should be a scheduler that will look into incoming packages and based on that adjust some timers?

s0170071, 2018-02-22:
@TD-er alright, then I get to it. I will also implement a small memory level history list to identify the calls that lead to an all-time memory low.

But can we postpone the reboot - timing - subtask restructuring a bit, please?  I would really like to tackle one memory issue after the other.

TD-er, 2018-02-22:
I think I can find some time to postpone fixing things ;)

mattlward, 2018-02-23:
I do not like the idea of turning things off and remaining off thru a reboot.  Some of my units are very remote, if they fail to start it is a drive to get to them.  Other than code loads, I try not to reboot them.

s0170071, 2018-02-23:
I see what you mean. Since memory drops can also be caused by incoming traffic and bit flips, disabling things would void any chance of recovery through the watchdog. So better just take a note of that condition and reboot and display an out-of-mem- reboot counter.
But what about users that overload / misconfigure their device? They would be stuck in a boot loop.

mattlward, 2018-02-23:
Interesting concern...  I have not hit overload on any of mine yet.  I did just misconfig one hand had to erase flash to bring it back.  But, that was done before deployment.  Maybe drop things after X numbers of reboots over Y amount of time?

Grovkillen, 2018-03-16:
Closing time?

uzi18, 2018-03-16:
I think we can add some information for user if freemem is low, about
system could be unstable from this moment.
We can also disable edit button on empty tasks, to not  going worst.

closed 2018-03-22 by mvdbro