letscontrolit/ESPEasy #924: Multiple Switch Inputs & Controller fails

jannnfe, 2018-02-22:
### Steps to reproduce
How can we trigger this problem?
Two Switch Inputs with different Names, with different GPIO on one ESP. Domoticz controller set and both inputs are assigned to another Domoticz IDX. With "Inversed Logic" turned on, the status of each switch is not updated. With "Inverse Logic" switched off, the status changes, but this also changes the status of the second switch input (Also see: [here](https://github.com/letscontrolit/ESPEasy/issues/673)) With Domoticz Controller turned off everything works as it should.

Does the problem presist after powering off and on?
YES!

### Expected behavior
Both switch inputs should change their status independently of each other. Both should independently publish their status via the Domoticz Controller.

### Actual behavior
The first switch input switches on the second switch input. With "Inversed Logic" turned on, the status of each switch is not updated. With "Inverse Logic" switched off, the status changes, but this also changes the status of the second switch input.

### System configuration
Hardware:
Wemos D1 Mini
Two Switch Inputs

Software or git version:
Tested all version above v2.0-20180217

TD-er, 2018-02-22:
Hmm without looking at the code, it sounds like some variable which is
shared between both instances of the plugin.
Sounds like something we can fix :)
What sounds a bit strange is that the domoticz controller affects behavior.
Is it the MQTT or the HTTP controller?
And does it also behave like this when only one of the switches is set to
the Domoticz controller?

---
Verzonden vanaf laptop

jannnfe, 2018-02-22:
Sorry for my incomplete statements. I only found the problem with the Domoticz MQTT controller.
I have also tested it now, if only one of the two switches sends inputs to the controller and the problem is the same.
[image]
Sensor1 should be State 0 and Sensor2 should be State 1 but they both switch the same, which is physically not possible, because one sensor is attached to the top of my garage door and the other below.

jannnfe, 2018-02-22:
Addendum: On another ESP8266 I only have one Switch Input and "Inversed Logic" off, everything works as it should. However, when I turn on "Inversed Logic" on this board sending to Domoticz is buggy. Logically, there is no problem with multiple switch inputs on this board, because there is only one Switch Input. Same Version, Same Board, Same GPIO.

TD-er, 2018-02-22:
You mention 

> Tested all version above v2.0-20180217

Does that mean you only tested these versions, or that was the first where the bug occurred and previous versions were OK?

jannnfe, 2018-02-22:
I have tested this issue on Version v2.0-20180217 and on the newest release of mega and v2.0. Sorry for my wrong mode of expression.

TD-er, 2018-02-22:
It's not wrong, nor incorrect.
Just wanted to make sure, since there was nothing I could think of that has been changed in the last week, which could cause anything like this.
I did change something related to the inverse state of a switch, but that's about 2 months ago I guess.

jannnfe, 2018-02-22:
Okay no problem. Maybe you find the problem and could look for it. Sorry if I have a lot of problems but I am currently switching my whole home system to Domoticz and that issue came up now. Please inform me if you know something new. Already thank you in advance!

Oxyandy, 2018-04-16:
GPIO showing wrong state with Domoticz MQTT enabled, 
hmm is this related to the issue I have been back-tracking #1216 ?

Grovkillen, 2018-10-23:
What's the status on this one?

TD-er, 2021-01-02:
This one should also be fixed by recent changes, please let me know if it is still a problem

closed 2021-01-02 by TD-er