letscontrolit/ESPEasy #926: Feature Request:  ADS1115 plug-in enhancement

Budman1758, 2018-02-23:
Would it be at all possible to re-engineer the ADS1115 plugin so that all four inputs are available from within a single task?  As it is now you have to create a separate task for each input and for setups that need a lot of different tasks this would save up to 3.  I'm working on an irrigation controller and it is designed for a lot more inputs than are actually currently possible.  (due to task limits)  My hope is down the road that will change. The idea for lots of inputs is to move to "on demand" watering vs a timer system.  This would be possible due to multiple soil moisture sensors. 

There are lots of examples of plugins that have more than 1 value possible so it seems like this would not be too extremely difficult.  Sure would go a long way toward saving task spots for other stuff. :)  

Thoughts?  I can supply the pizza and suds... :)

enesbcs, 2018-02-24:
I've a similar, simple solution for ADS1015:
https://github.com/enesbcs/ESPEasyPluginPlayground/blob/master/_P167_ADS1015.ino

wclintdavis, 2022-09-17:
This would be very helpful to me also, I'm sure others would like this as well

PR #4786 (2023-09-06): [ADS1x15] Support ADS1015 & add sample rate selector
Make timeout depending on set sample rate + add sample rate selector

ToDo:
- [x] Document
- [x] Show detected version (ADS1015/ADS1115)
- [x] Add option for multiple taskvalues per task

Fixes: #4768 
Fixes: #1677
Fixes: #1564 
Fixes: #926
closed 2023-09-06 by TD-er