letscontrolit/ESPEasy #930: rename SoftwareSerial lib

uzi18, 2018-02-23:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

please rename SoftwareSerial lib dir to ESPeasySoftwareSerial, it is confusing some builders.

uzi18, 2018-02-23:
@TD-er @Grovkillen is it possible?

Grovkillen, 2018-02-23:
Sure, I can agree.

TD-er, 2018-02-23:
Sounds good to me.

uzi18, 2018-02-24:
Ok I will try to prepare PR.

TD-er, 2018-02-24:
Please do in V2.0

uzi18, 2018-02-24:
ok, but there is no ESPeasySoftwareSerial in v2.0

TD-er, 2018-02-24:
Ah I see, you're right.
Now I am near a computer instead of replying from a phone, so I don't have to recollect everything from memory ;)

uzi18, 2018-02-24:
You want to move all SoftwareSerial plugins to ESPSoftwareSerial?
Does it support multiple serial?

TD-er, 2018-02-24:
It supports multiple serial.
In fact it is the same library, only with a few minor patches from me.

- Max 3 simultaneous uses (can be changed easily in one line)
- Less memory use (due to the max limit of 3)
- Virtual destructor.

closed 2018-02-27 by TD-er