letscontrolit/ESPEasy #938: login page works wrong

uzi18, 2018-02-24:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
set password, restart espeasy

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes

### Actual behavior
unable to input password

### System configuration
mega branch from yesterday

uzi18, 2018-02-24:
[image]

html code:
`<!DOCTYPE html><html lang='en'><head><meta charset='utf-8'/><meta name='viewport' content='width=device-width, initial-scale=1.0'><title>ESP_Easy</title><style>* {font-family:sans-serif; font-size:12pt;}h1 {font-size:16pt; color:#07D; margin:8px 0; font-weight:bold;}h2 {font-size:12pt; margin:0 -4px; padding:6px; background-color:#444; color:#FFF; font-weight:bold;}h3 {font-size:12pt; margin:16px -4px 0 -4px; padding:4px; background-color:#EEE; color:#444; font-weight:bold;}h6 {font-size:10pt; color:#07D;}.button {margin:4px; padding:4px 16px; background-color:#07D; color:#FFF; text-decoration:none; border-radius:4px}.button.link {}.button.help {padding:2px 4px; border:solid 1px #FFF; border-radius:50%}.button:hover {background:#369;}th {padding:6px; background-color:#444; color:#FFF; border-color:#888; font-weight:bold;}td {padding:4px;}tr {padding:4px;}table {color:#000;}.note {color:#444; font-style:italic}.headermenu {position:fixed; top:0; left:0; right:0; height:64px; padding:8px 12px; background-color:#F8F8F8; border-bottom: 1px solid #DDD;}.bodymenu {margin-top:96px;}.menubar {position:inherit; top:44px;}.menu {float:left; height:20px; padding: 4px 16px 8px 16px; color:#444; white-space:nowrap; border:solid transparent; border-width: 4px 1px 1px; border-radius: 4px 4px 0 0; text-decoration: none;}.menu.active {color:#000; background-color:#FFF; border-color:#07D #DDD #FFF;}.menu:hover {color:#000; background:#DEF;}.on {color:green;}.off {color:red;}.div_l {float:left;}.div_r {float:right; margin:2px; padding:1px 10px; border-radius:4px; background-color:#080; color:white;}.div_br {clear:both;}.alert {padding: 20px; background-color: #f44336; color: white; margin-bottom: 15px;}.closebtn {margin-left: 15px; color: white; font-weight: bold; float: right; font-size: 22px; line-height: 20px; cursor: pointer; transition: 0.3s;}.closebtn:hover {color: black;}</style></head><body><header class='headermenu'><h1>ESP Easy Mega: ESP_Easy</h1></header><section><form method='post'><table><TR><TD>Password<TD><input type='password' name='password' value=''><TR><TD><TD><input class='button link' type='submit' value='Submit'><TR><TD></table></form></section><footer><h6>Powered by www.letscontrolit.com</h6></footer></body>
`

uzi18, 2018-02-24:
password input is under header

TD-er, 2018-02-25:
Very quick solution to allow login, until a proper fix is ready:

Right mouseclick on webpage and select "inspect":
[image]

Navigate to the header part of the page and look for the ".headermenu" section in the CSS:
[image]
Depending on where you clicked in the webinterface, it may already be highlighted.

Deselect "position: fixed;"
[image]

Then the login part will appear to allow login.
[image]

You may want to temporarily remove the password, until the CSS is properly fixed in the next build.

uzi18, 2018-02-25:
You can add password to url address:   
`/login?password=something`

closed 2018-03-09 by TD-er